cleanflight/cleanflight #531: Add receiver calibration

digitalentity, 2015-02-20:
I have a flawed transmitter with one channel (roll) ranging from 1250 to 1700 instead of a normal 1000-2000 range. Also no transmitter is perfect and each has a slightly different range and midpoint.

Please consider adding a receiver calibration configuration to cleanflight to be able to scale actual transmitter range to expected 1000-2000 range.

digitalentity, 2015-02-20:
I have modified CF code to remap rc input from custom range to normal [1000;2000] range right after channel remap. Also added a new CLI command 'calibration' to manage the configuration. Unfortunately new configuration is incompatible with previous ones.

TODO: do extensive testing; include the rc calibration config to cli dump

matbogdan79, 2015-02-20:
With a faulty potentiometer on your transmiter I won't recommend flying drones.
Buy a Taranis, that have gimbal calibration.

digitalentity, 2015-02-20:
@matbogdan79 the potentiometer is not faulty, it is wrong angle (120 degree instead of needed 60 deg). Other than that transmitter is working flawlessly. I am not going to trash a working piece of equipment and spend money on a new one rather than have a flight controller with software capability of compensating for this flaw.
OpenPilot can do this, Taulabs can do this, NAZA can do this, why CF can't?

digitalentity, 2015-02-22:
Today I flew a quad with modified firmware and a transmitter with flawed roll axis range. Didn't notice any difference from an all-good transmitter. Currenltly there is no way to access and modify calibration data except for CLI and I believe there is currently no need to make one. @hydra, do you think this feature is worth merging to master? I'll make some cleanups to my code and with your approval will create a pull request.

hydra, 2015-02-22:
Please create a pull request - i can see it being useful for people that have old transmitter/receiver combinations.

digitalentity, 2015-02-23:
Created a pull request #542. Also did a fix to prevent calibration from messing up with throttle and other channel values while in failsafe mode.

digitalentity, 2015-03-05:
@hydra sorry for bothering, I've made cleanups to the code and rebased #542 to most recent 'master' so it could be cleanly merged.

digitalentity, 2015-04-03:
Pull request updated (#618) to cleanly merge with 1.8.1

digitalentity, 2015-09-28:
Resolved and merged into 1.10

closed 2015-09-28 by digitalentity