cleanflight/cleanflight #532: [Idea] Onboard SD card

projectkk2glider, 2015-02-20:
The Flip32 board could be easily modified to support direct connection to an SD card (this is probably also true for Naze32 and other clones). This would then give a possibility to implement black-box logging on-board without the need for external serial logging device. This should also support higher logging rates (more RAM for write buffer).

The hardware wiring on Flip32:
* Use SPI1 peripheral
* SCK pin PA5 (pin 15) is wired to pin 9 of MMA8452Q accelerometer (which is not fitted on board). This is the only **hard** pin to reach, ti will require precise soldering, but the pin is quite acessible.
* MISO pin PA6 (pin16) which is wired on RC input 5
* MOSI pin PA7 (pin17) which is wired on RC input 6.
* 3.3V VCC 
* GND

Since RC5 and RC6 are used by softserial 1, we loose that functionality, but gain SD card access. What do you think?

hydra, 2015-02-20:
The problem is the filesystem.  this has been discussed before on irc.  with a flash chip or serial no filesystem support is needed in the flight controller software.  A library would be needed for such support.

If someone wants to try it please give it a go and maybe link to your code from here so we can see how it pans out?

thenickdude, 2015-02-20:
@tommyleo achieved this with his port of Baseflight to the [μBrain's 168MHz STM32 CPU](https://vrbrain.wordpress.com/vr-microbrain/), called [speedyflight](https://github.com/tommyleo/speedyflight). He used [fatfs](http://elm-chan.org/fsw/ff/00index_e.html) as the FAT filesystem driver.

[fatfs takes 4.6-7.1 kB of flash](http://elm-chan.org/fsw/ff/en/appnote.html) depending on selected features, if using ARM7 Thumb instructions.

BUT as far I can tell, fatfs is completely based around blocking operations: Writing a single character to the SD card can cause a flush that [must wait](https://github.com/tommyleo/speedyflight/blob/master/lib/fatfs/fatfs_sd.c#L129) for the card to become ready. This can take 150ms on slow cards, and the flight control loop will be paused until that completes. By default it even uses a 500ms timeout, so I imagine if the SD card falls out during flight, terrible things will happen.

By contrast, my [flashfs](https://github.com/sherlockflight/cleanflight-dev/blob/blackbox-flash/src/main/io/flashfs.c) for writing to NOR flash never waits for the flash device to become ready, if it is busy then data is buffered instead, and if the buffer is overrun then data is dropped rather than freezing the flight controller for many milliseconds. 

A FAT filesystem driver which supports asynchronous writes like this would seem like a prerequisite, but I don't know of one. Alternatively we could adapt flashfs to write to an SD card with no filesystem pretty easily, but end users wouldn't be able to read the cards without a lot of pain on their part.

handsomejackuk, 2015-02-21:
i would love this too... i see there i2c card reasers / writers. please escuse me too many beers... mini camera module and integral camera would be brillant...also for data logging

tommyleo, 2015-02-21:
Hi all,

for first my compliment to @thenickdude for the blackbox code and to @hydra and the comunity for Cleanflight!

I use <a href="http://stm32f4-discovery.com/2014/07/library-21-read-sd-card-fatfs-stm32f4xx-devices/">Libraries and tutorials for STM32F4 series MCUs by Tilen Majerle</a> 

I think that it is very easy to insert some check to prevent dangerous latency and insert some controls for safety(set a maximum time in µs to complete the write process)!

In my porting(I start from Baseflight) it is too hard for me to port Cleanflight, I don't have so much experience on STM32F4, for me it is just an hobby and I don't have so much time, but the BlackBox work very well on my porting without doing change to the origal library!

I try with looptime=1000 and the variance is not up 5µs! :)

In my porting looptime = 1000 is very stable, with this sensors: MPU6000 (SPI Gyro + Acc), HMC5883 (I2C External Compass), MS5611 (SPI Barometer), GPS Ublox, SBUS receiver and SDCARD(SPI) + BlackBox!

I have also try to remove the card when the system working and not problem, clearly you loose the logs but not problem for the system!

p.s. I will wait this Cleanflight milestones "STMF4 Targets"  :)

enly1, 2015-02-21:
> I use Libraries and tutorials for STM32F4 series MCUs by Tilen Majerle

Not sure if anmything matters here, but my data logger that I developed on
STmF103, I had to worry about the I2C and SPI peripheral clashes when I did
stuff as they shared the same DMA channel.  The write time on the SPI / SD
card was too long to accommodate in my polling rate on I2C, so they HAD to
be the right combo of I2C and SPI devices being used.

tommyleo, 2015-02-21:
I think that I'm lucky, STM32F4 is another story!

On board μBrain I work with a dedicated SPI3 channel for SDCard. The board can work with looptime like 200μs / 500μs very stable, with many feature and BlackBox it is better 1000μs, 1Khz cycles time is enough.

I have also a flip32 acro(without compass/baro/gps) to compare and with 1600 looptime, I see latency about 50/100/200μs...so also the feature BlackBox it's not so stable!

thenickdude, 2015-02-21:
@tommyleo, try filling the SD card with files, then delete them, and retest. That should trigger the worst-case behaviour.

hydra, 2015-02-22:
yes - you need to trigger erase cycles.

tommyleo, 2015-02-27:
Hello everyone,

I had a busy week at work.

Right now I did the test, I filled the SD Card with 16GB and after I did a delete of all files.

I made a flight, with this sensors: gyro, acc, mag, baro, GPS 

Looptime = 1000

This is the log: <a href="https://www.dropbox.com/s/klfmja2rigwbsb7/LOG00005.TXT?dl=0">Log File<a>

thenickdude, 2015-03-01:
The longest observed looptimes from your log:

```
1835	1791
1834	1791
1831	1790
1830	1789
1826	1787
1821	1785
1821	1785
1820	1785
1818	1783
1817	1783
1816	1779
1812	1777
1810	1560
1810	1553
1809	1552
1808	1547
1808	1545
1808	1545
1807	1541
1807	1535
1807	1534
1806	1533
1805	1533
1805	1269
1804	1261
1804	1260
1803	1260
1801	1242
1800	1055
1799	1055
1799	1054
1798	1052
1797	1009
1796	1008
1795	1008
1795	1008
1795	1007
1794	1007
1794	1007
```

That's actually shockingly good performance considering how slow microSD cards can be without violating the spec. That's a Sandisk Extreme, right?

tommyleo, 2015-03-01:
Yes, Sandisk Extreme 16GB Class 10 UHS-I microSDHC

I think that is good performance, considering looptime=1000 and rate 1/1, and all sensors activated!

closed 2017-02-25 by hydra