cleanflight/cleanflight #540: will not go into Autotune Mode !

mdmanmike, 2015-02-22:
Hi first off I love the CleanFlight version! I cannot get it to switch into AutoTune Mode. I have an Afro Naze32, CleanFlight 1.7.2, PID Controller 1(Rewrite). 2 pos. Switch set to Horizon and AutoTune. all other modes disabled in CleanFlight, FrSky Taranis Plus with FrSky X8R. I arm motors with a Switch not stick. I turn on Radio, attached Lipo,Arm Motors (in Horizon Mode). take off and stabilize quad, when stabile I switch to AutoTune .....and nothing, no tuning no nothing its like flying in Horz mode. I have checked this several times in MODES in CleanFlight and when I switch my switch into AutoTune it lights the AutoTune Icon, so I know that it's switching into AutoTune but not Tuning. My PID's have not changed either. Any Ideas ? Thanks
Mike

matbogdan79, 2015-02-22:
activat the autotune from CLI or interface than use aux to activate it in flight. Make sure you only autotune it for a few seconds (2-3 sec) or it will get worst.

tricopterY, 2015-02-22:
Change your PID Controller to 0 or 3, for auto-tune. Rewrite is not supported.

mdmanmike, 2015-02-22:
Thanks I'll give it a try and let you know.

mdmanmike, 2015-02-22:
Thanks Mat, autotune IS activated in CLI no go. I'll try  tricopter's suggestion and set PID Controller to "0"

tricopterY, 2015-02-22:
I just checked the source code 'Rewrite' is also supported, only the PID Controllers that do floating-point arithmetic are not supported.. So, not sure why it is not working for you.

hydra, 2015-02-22:
This issue tracker is for bug reports, not support.  Please try posting in the forums or ask on the IRC channel first.  If there's still an issue please re-open this issue and give full details of how to replicate the problem and configuration dumps.

closed 2015-02-22 by hydra

mdmanmike, 2015-02-23:
I thought this is a bug since auto tune worked with ver. 1.7.1 and not in 1.7.2, hmmm I must be mistaken.

hydra, 2015-02-23:
Please check the motor outputs tab. You should see activity when in auto tune mode when armed and in auto tune mode.