cleanflight/cleanflight #541: Disarm causes Failsafe Mode to trigger

RandGH, 2015-02-22:
Sometimes (excat conditions unknown) setting the disarm switch to disarm causes the board to activate failsafe (or at least start motor in failsafe speed).
Naze32 Full, CF 1.7.2 (Feb 12 2015 13:40:16)
Up until know i only noticed when connected to CF GUI, i cant make it behave this way. 
I've captured a set of debugging info, maybe it helps

And why cant i attach a txt/csv file?;)
"Unfortunately, we don't support that file type. Try again with a PNG, GIF, or JPG. "
I'll add as plain text in second post

RandGH, 2015-02-22:
Triggered issue 3 times , Arm is on Aux1 (Disarm, Horizon/Arm, Rate/Arm)

timestamp,gyroscopeX,gyroscopeY,gyroscopeZ,accelerometerX,accelerometerY,accelerometerZ,magnetometerX,magnetometerY,magnetometerZ,RC0,RC1,RC2,RC3,RC4,RC5,RC6,RC7,RC8,RC9,RC10,RC11,RC12,RC13,RC14,RC15,Motor0,Motor1,Motor2,Motor3,Motor4,Motor5,Motor6,Motor7,Debug0,Debug1,Debug2,Debug3
1424641643710,-0.48780487804878053,-0.24390243902439027,-0.24390243902439027,-0.001953125,0.111328125,1.00390625,-1.343119266055046,0.14311926605504588,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-137
1424641643811,-0.48780487804878053,0.24390243902439027,-0.24390243902439027,-0.001953125,0.109375,1.001953125,-1.3412844036697247,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1163,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,-2,-137
1424641643912,-0.7317073170731708,0.24390243902439027,-0.24390243902439027,-0.001953125,0.111328125,1.00390625,-1.3422018348623854,0.14311926605504588,-3.757798165137615,1520,1521,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-137
1424641644013,-0.48780487804878053,0.24390243902439027,-0.24390243902439027,-0.00390625,0.109375,1.001953125,-1.3440366972477065,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-137
1424641644114,-0.48780487804878053,0.24390243902439027,-0.24390243902439027,-0.001953125,0.111328125,1.00390625,-1.346788990825688,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-136
1424641644215,-0.48780487804878053,-0.24390243902439027,-0.24390243902439027,-0.001953125,0.111328125,1,-1.3458715596330275,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-136
1424641644316,-0.48780487804878053,0,-0.24390243902439027,-0.00390625,0.111328125,1.00390625,-1.3403669724770642,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,2,-136
1424641644418,-0.48780487804878053,0,-0.24390243902439027,-0.00390625,0.111328125,1.001953125,-1.3440366972477065,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-135
1424641644518,-0.7317073170731708,-0.24390243902439027,-0.24390243902439027,-0.001953125,0.109375,1.001953125,-1.343119266055046,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-135
1424641644620,-0.48780487804878053,-0.48780487804878053,0,-0.001953125,0.109375,1.001953125,-1.3440366972477065,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-135
1424641644722,-0.48780487804878053,0,0,-0.001953125,0.111328125,1.001953125,-1.343119266055046,0.14495412844036698,-3.757798165137615,1520,1520,1516,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-134
1424641644824,-0.48780487804878053,-0.24390243902439027,0,-0.00390625,0.109375,1.001953125,-1.3440366972477065,0.14403669724770643,-3.757798165137615,1520,1520,1516,1010,1520,1043,1008,1043,1163,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-134
1424641644924,-0.48780487804878053,0,-0.24390243902439027,-0.00390625,0.109375,1.00390625,-1.3477064220183486,0.14311926605504588,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-134
1424641645026,-0.24390243902439027,0,0,-0.001953125,0.107421875,1.00390625,-1.3458715596330275,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1163,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-134
1424641645127,-0.48780487804878053,0,0,-0.001953125,0.109375,1.00390625,-1.3422018348623854,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-134
1424641645228,-0.48780487804878053,0.24390243902439027,0,-0.001953125,0.109375,1.001953125,-1.3412844036697247,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-134
1424641645329,-0.48780487804878053,0.24390243902439027,0,-0.001953125,0.111328125,1.00390625,-1.344954128440367,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-133
1424641645431,-0.24390243902439027,0,0,-0.001953125,0.109375,1.001953125,-1.3458715596330275,0.14495412844036698,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,0,-133
1424641645533,-0.48780487804878053,0.24390243902439027,-0.24390243902439027,-0.001953125,0.109375,1.001953125,-1.346788990825688,0.14403669724770643,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520, [... cut ...]
1424641665469,-0.48780487804878053,0,-0.24390243902439027,-0.00390625,0.119140625,1,-1.3422018348623854,0.14036697247706423,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1165,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,-1,-177
1424641665570,-0.9756097560975611,0.24390243902439027,-0.24390243902439027,-0.001953125,0.12109375,1.005859375,-1.3440366972477065,0.14036697247706423,-3.757798165137615,1520,1520,1517,1010,1520,1043,1008,1043,1163,1520,1520,1520,1520,1520,1520,1520,1000,1000,1000,1000,0,0,0,0,0,0,1,-176

thenickdude, 2015-02-22:
Check that your channels on the receiver tab don't fall below the default failsafe_min_usec of 985 or above the failsafe_max_usec of 2115. Also run the "dump" command on the CLI, paste the resulting config to http://pastebin.com/ and post the link here.

RandGH, 2015-02-22:
They dont.
Upgraded to the Feb 20 build now, same issue occured once more; this time I noticed that the sliders in the motor tab where set to values (different ones this time).
Is that adjustable via Sticks? Maybe I am unwittingly adjusting a value here?

http://pastebin.com/B8TFggyf

thenickdude, 2015-02-22:
Do you have an OSD connected to your main serial port?

tricopterY, 2015-02-22:
He is using PID Controller 'Harikiri', all values are in floating-point.

thenickdude, 2015-02-23:
Really? I'm looking at the code for the Harakiri PID controller now and it reads pidProfile->P8[axis], not any of the floating point PIDs.

tricopterY, 2015-02-23:
float PTerm, ITerm, DTerm, PTermACC = 0.0f, ITermACC = 0.0f, ITermGYRO, error, prop = 0.0f; #L 526 
PTermACC = error * (float)pidProfile->P8[PIDLEVEL] * 0.008f; #L 557

thenickdude, 2015-02-23:
Yup that's the line I'm talking about. It reads P8[] which is an integer PID of type uint8_t published as `p_pitch` in the CLI, not P_f[] which is the equivalent floating point version of the PID setting which is only used by the pidLuxFloat controller and published as `p_pitchf` etc in the CLI.

tricopterY, 2015-02-23:
But isn't PIDLEVEL an enum to float in pidProfile_t?

thenickdude, 2015-02-23:
It's an index into the P8 array, which is an array of uint8_t PIDs. It doesn't access any of the floating point PIDs, to do that the code would have to reference P_f[], I_f[], D_f[], A_level, H_level, or H_sensitivity.

tricopterY, 2015-02-23:
Yup - sorry got it. I am just looking around and familiarising myself. So, how does he fly with P=0? Does that mean Open Loop?

RandGH, 2015-02-23:
Sorry for the confusion, I manually set all values to 0 as I am trying to tune it with a copy of Nebbians DynonTunig rig.
While trying to do that I noticed the issue.
And I have not set any values on the motor tab manually, that's why I noted it.
And no, I have not set failsafe on the tx at all.

hydra, 2015-02-23:
There is a bug with the checkbox on the motors tab - if you can find the exact sequence that causes it to show a checked value when it should not be checked please document the steps to repeat an open an issue on the cleanflight-configurator issue tracker (not this one).

Avoid switching to/from the motors tab while armed for now.

Maybe there's some combination of settings you can find that highlights the problem but likely it's the configurator setting motor speeds when the checkbox is checked.

Can you reset to defaults and provide steps to repeat and re-open the ticket if there's still an issue?

closed 2015-02-23 by hydra

RandGH, 2015-02-26:
See cf configurator ticket for steps to reproduce

brrc, 2016-03-01:
How do you think it's the best way to overcome this, should it not be possible to turn on the test mode, or simply disarm if they are armed?

borisbstyle, 2016-03-01:
check your modes tab in configurator and see for failsafe mode