cleanflight/cleanflight #544: inflight adjustment work on all profiles

handsomejackuk, 2015-02-23:
Hi, is there any  known issue with inflight adjustments working across all profiles even though the adjustment is only enabled on profile 2... version 1.7.1 cleanflight naze32...  slot 1 ? i dont really understand the slot 1 is this relevant...?

i have profile 2 setup on aux 1 a pot on my turnigy 9xr to adjust rate pitch P adjustment and if i switch profiles to profile 1 or profile 3 the pot still adjusts pitch roll P gain.... do i need to reboot flight controller between switching profiles ?

hydra, 2015-02-26:
As a workaround, define an adjustment in the other two profiles that uses the same aux channel and same slot in the other two profiles but set it to 'do nothing' / 'no changes'.

I think the adjustment state needs resetting when profiles are switched.  This is a bug, thanks for the report.

hydra, 2015-02-26:
That commit /should/ fix it, please re-open once you have verified the fix using a binary that includes this change.

commit 55d9994f (2015-02-26): Reset adjustment states when switching profiles.

Fixes #544.
closed 2015-02-26 by hydra