cleanflight/cleanflight #545: CC3D with sensor board GY-86 dont work correctly

natanke, 2015-02-24:
I have CC3D with CF 1.7.2 and sensor board GY-86(10dof) Baro MS5611 start to work only when checkbox Display is active. Mag sensor HMC5883L still dont work.
All sensors on this board are operational.
[image]

digitalentity, 2015-02-24:
HMC5883L on GY-86 board won't currently work with CF. This board has HMC5883L comapss connected to MPU6050 secondary I2C interface, which is not accesible directly without initialising MPU6050 first. However, barometer should work with latest build (1.8.0 RC).

ledvinap, 2015-02-24:
It should be quite trivial to fix. Setting I2C_BYPASS_EN should be enough.

digitalentity, 2015-02-24:
This whould require detecting and initializing MPU6050 first. As fas as I know currently CF does not detect acc/gyro on external I2C interfaces.

ledvinap, 2015-02-24:
It may be possible to use external MPU6050 and ignore onboard MPU6000. In fact this seems to be default (mpu6050 is checked first), just enable MPU6050 on CC3D target ...

hydra, 2015-02-24:
This all seems rather pointless though, just buy a board with the sensors you want to use on it.

However, the sensor code was cleaned up recently and it would perhaps be nice to be able to choose exactly what hardware to use if you have got lots of hardware connected.  The changes are trivial to enable this now.

points you could change are here (to allow the user to choose each sensor)

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/initialisation.c#L554
https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/initialisation.c#L565
https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/initialisation.c#L561

https://github.com/cleanflight/cleanflight/blob/master/src/main/config/config_master.h#L46
https://github.com/cleanflight/cleanflight/blob/master/src/main/config/config_master.h#L53

then use the same retry mechanism as found here:

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/initialisation.c#L527-L531

I2C would also need enabling appropriately.  IIRC On the CC3D I2C is enabled by default unless you are using the serial port that is shared with the I2C bus pins.

you can see this here:

https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L281-L284

Enabling MPU6050 bypass is done here:

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu6050.c#L346-L347

I would welcome a PR that makes these changes, I'm busy with other tasks myself right now so if you want it you'll have to do it :smile:

hydra, 2015-02-24:
I'm not aware of any code that couples i2c display initialisation with baro initialisation (and rightly so).

See

https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L276-L288

and

https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L310-L314

followed by

https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L316-L319

It's possible that enabling the display feature introduces a delay which isn't normally present and perhaps your board requires a small delay to allow the baro to finish initialising after power up before it is able to respond.

Maybe try adding disabling the 'DISPLAY' feature and adding some delay() commands before the baro is detected, here:

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/initialisation.c#L398

Please report your findings.

hydra, 2015-03-02:
@natanke Any update?

natanke, 2015-03-03:
@hydra No, because I'm not a programmer.

hydra, 2015-03-03:
Ahh OK, in that case perhaps I or someone else will make you a binary to test with the extra delay in it as above when we get time.

natanke, 2015-03-03:
@hydra well thank you, I'll wait to test it)

rebdel, 2015-03-03:
Hi, i´m also interrested.. which piece of code must be editied?

Can compile it myself..

thx

hodaddy4, 2015-03-24:
Any progress on this? I'm more of a hardware than a software guy, but I'm learning and I have same sensors sitting around that I got without knowing what I was getting into

hydra, 2015-05-15:
@rebdel @hodaddy4 binaries are available here http://cleanflight.memoryleaks.org/builds/

Please test and re-open if there is still an issue.

closed 2015-05-15 by hydra

rebdel, 2015-05-16:
Hi Dominic,

thx.. tested it with latest build (Cleanflight/CC3D 1.9.0 May 15 2015 / 11:16:34 (dee68f5), but only Baro will be recognized after display enabling.

Can someone confirm ?
BR
Frank

natanke, 2015-05-16:
I confirm, with latest build works only baro.