cleanflight/cleanflight #546: Improve Oneshot ESC startup behaviour

olexs, 2015-02-24:
ESC initialization at bootup with Oneshot enabled happens with weird artefacts: ESCs not initializing in sync, and some ESCs initializing twice. Video of async and double init sequence beeps (using BLHeli 13.0 on HK BS12A ESCs with Oneshot enabled): https://www.dropbox.com/s/v87p3ttz066dbks/2015-02-24%2016.33.47.mp4?dl=0

Probably linked to 500ms blocking LED blink code in the boot sequence: https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L323-L334

hydra, 2015-02-26:
also as reported switching profiles also makes some BLHeli oneshot escs re-initialise.  Likely both caused by signal loss at the ESCs during beeping events.

The buzzer/beep/notification system needs an overhaul.  Generating beeps should not block.

PaulHam211, 2015-02-26:
Dominic that might be the reason. 
I reported this on the irc channel the other day. I didn't realise at the time that the escs also re initalise when trimming the accelerometer using the sticks

fiveangle, 2015-02-26:
FYI: I've not seen any of these issues on Flyduino PR1575 K.I.S.S. ESC with Oneshot125 enabled so it may be related to BLHeli specifically and not Oneshot125 ESCs in general.

-=dave

Ravenholm14, 2015-03-08:
The double Int beeps in BLHeli 13 is a documented feature not a bug. Because the ESC F/W can run either with or without Oneshot, he second set of beeps are to signify it has detected Oneshot and has enabled it in the ESC F/W. Thus why you don't get it with the KISS ESC's, it's not a Oneshot problem but a BLHeli Feature.

anthonycake, 2015-03-16:
I have captured the startup sequence on a scope, and observe the following: 
Prior to the restart, the Oneshot period is 3.5ms, corresponding to the configured loop time
During restart, the Oneshot disappears, cleanly, for approx. 900ms
After restart, Oneshot restarts, but with a period of ~8.2ms, for a duration of approx. 300ms. 
After 300ms, the Oneshot period drops to 3.5ms again. 

The issue that we faced was that the ESCs had an internal timer, which was wrapping at 4ms. During restart, we had a dangerous condition where the motors would start for about 1/2 second, long enough to do some damage if props were left attached. 

My suggestion would be to only start emitting OneShot when the main loop starts.

ledvinap, 2015-03-16:
Maybe setting initial counter value to 0 in https://github.com/cleanflight/cleanflight/blob/2a73ad69253be498611b79b38c8f6bb52990f774/src/main/drivers/pwm_output.c#L181 may help.
```
motors[motorIndex] = pwmOutConfig(timerHardware, ONESHOT125_TIMER_MHZ, 0xFFFF, 0);
```

ledvinap, 2015-03-16:
But it won't fix it completely ... loop() is not guaranteed to execute at looptime and there are delays, mainly when unarmed.

yGO77, 2015-04-15:
With last cleanflight firmware ( 1.8.1 ) I have two motors that spin when cleanflight reboots. 
here what I observed so far:

[image]

It looks like in the case where the motor is spinning a pulse of 143 us is sent right after the restart, then  all the next pulses are 125 us.

In the case the motor does not spin, the 1st pulse is 119 us, then all the next pulses are 125 us.

Hope this will help.

hydra, 2015-04-22:
Please re-test now that #730 is merged.

hydra, 2015-04-23:
Ignore previous comment, #730 didn't change the startup beeps to use the new beeper system, that still needs doing.

hydra, 2015-05-03:
@PaulHam211 can you re-test using master as the other beeper changes should have fixed the blheli esc initialisation issues when changing profiles, trimming acc, etc.

hydra, 2015-05-29:
Should be fixed by the following commits:

0c1a6c5c2fbe6d6d4b649744ba876d9e103156f9 22a98af25a452fb9bead2d218338a395dd93d312 e6733b4dfccf8a32c297f8ccdd214ae3e1619a55

Please test and report back ASAP!

gunadeau, 2015-06-03:
Hello, I just flashed my Naze32 withe the new version of cleanflight 1.9.0 (stable) 2015-06-02  00:41

I'am wondering if Oneshot get activated on my ESC's at booting. On Cleanflight version 1.8.1 my ESC's was doing two set of beep at booting when Oneshot125 was enabled. But now it only doing one set of beep on both configuration Oneshot active and oneshot not active.

I have rotorgeeks ESC's flashed with BLHELI 13.1

I also tried BLHELI 13.2 and same thing.

Havoc256, 2015-06-04:
I am also seeing the same issues that gunadeau is reporting

I have the rotorgeeks 12amp flashed with BLHeli 13.1  Guess I have to go back to 1.8.1 for now.

gunadeau, 2015-06-10:
Here is an update from rotorgeeks.com about their ESC's:
Update - Cleanflight 1.9.0 or later does not issue the second beeps but you don't need them for confirmation.  If you have the Oneshot box checked in Cleanflight and your motors spin, you are in Oneshot mode.  When thix box is checked, the flight controller will only issue Oneshot protocol and the ESC cannot be calibrated to that range if they're aren't Oneshot capable.  So if it spins, it's working!

ProDrone, 2015-06-19:
@hydra Can be closed.

This issue was solved when PR #921 (System - Lock active features after init) was merged at:
https://github.com/cleanflight/cleanflight/commit/bbb86b30fa6da67eb8d29378ce119ab45691f2af

ProDrone, 2015-06-19:
Why Oneshot ESC's may produce additional beeps sometimes.

Explanation:

There is a difference in how ESC are controlled by normal PWM or Oneshot PWM as follows:

After starting up the flight controller:

a) ESC's using __normal PWM__ receive a __never interrupted__ PWM signal.
This is independend of looptime and other additional delays that may occur.
So it __beeps once__ on startup.

b) ESC's using __Oneshot PWM__ receive pulses that are __synchronized with the mainloop__.
Most of the time (for instance in flight) the mainloop is __constantly updating__ the Oneshot PWM pulses.

__However__, when some action is started that takes a bit longer, the mainloop has __no time to update__ the PWM pulses. This can happen when switching profiles or updating things using the Configurator or the CLI. On these occasions the EEPROM has to read or write all values, which takes a longer time.
After the interruption the __mainloop resumes updating__ the Oneshot pulses.
The ESC's will see this as a new __start of valid pulses__ and will __beep__ again.

hydra, 2015-06-24:
Thanks for feedback everyone.  Yes, the second set of beeps is gone from the motors, expected behavior.  That was an indication of the problem.

closed 2015-06-24 by hydra