cleanflight/cleanflight #547: FC crashes when using certain serial combo

danielolsman, 2015-02-24:
My flip32+ "crashes" and my ESCs are beeping slowly when using this serial configuration:

UART1 : MSP, CLI, TELEMETRY, SMARTPORT TELEMETRY, GPS-PASSTHROUGH
UART2: SerialRX (SBUS)
Softserial1:MSP
Softserial2:GPS

UART1 is connected to an bluetooth module, UART2 to a orangerx r620, Softserial1 to a MavlinkOSD 2.0 runnning MWOSD 1.3 and Softserial2 is connected to a Crius NEO6 GPS module.

All serial ports are set to 19200 baud and it works for a few minutes.
After a while my escs are beeping slowly and there is no MSP data out.

I tried a lower baud rate (9600), it still crashes.
Also tried to reload the firmware with the boot jumper shorted (FW 1.7.2)

stronnag, 2015-02-24:
orangerx r620 and sbus? No excuse to crash though.

stronnag, 2015-02-24:
Bench testing a similar combo (GPS on SS1, MSP (mwp) on SS2 both @19200 I certainly see large data loss (maybe 15% loss on SS2). Shall not to tempted to fly it.

hydra, 2015-02-24:
There are the incoming softserial changes to improve performance.  This is a known issue and other issues are already open. Closing as duplicate.

closed 2015-02-24 by hydra