cleanflight/cleanflight #548: Tricopter 3d Mode - Required Feature Missing - Naze

smarblewnec, 2015-02-24:
Hello,
Unfortunately I am not a programmer and so I am looking for help fixing this issue.  The issue is with tricopters in 3d mode using the Naze. The rudder servo needs to reverse when the throttle goes negative and currently the software does not allow for this. Due to the motor reversing the thrust angle on the rudder motor changes 180 degrees off from what the flight controller expects. Currently, once a tricopter is inverted it spins uncontrollably.  If this is not the correct place for this then please direct me somewhere else but this is a repeatable problem and will need code written to fix so I am hoping this is the right place to post this.
Thanks
Steve

hydra, 2015-02-26:
Thanks for the report.  Having it logged here is the best place, if you want to get someone to look into this then feel free to post a link to this issue along with your call for help.

nebbian, 2015-03-03:
I've made a code change to get this working, and it's working in bench testing.  Will submit a pull request when it's flight tested (thanks @smarblewnec)

catchthewind, 2015-03-03:
That would be great to have it working. 
How long is it going to take to have it implemented into official version?

smarblewnec, 2015-03-04:
I should be able to test fly it in the next couple of days.  After that, I am not sure how long it will take. So far the bench testing works great!   Nebbian may have more info on how long it will take to get into a firmware update.

catchthewind, 2015-03-04:
Great, thanks for your help!!

smarblewnec, 2015-03-05:
Nebbian, It works!  Thank you so much!  It flies inverted just like it should now.  I have lots of tuning to do before I really push it inverted but the servo issue is fixed!

catchthewind, 2015-03-05:
Is it already implemented or do we have to wait some time for 1.7.3 ?

smarblewnec, 2015-03-05:
You can find the test firmware at the link below so you can start using it right away.  Hydra or Nebbian may have more insight on when it would be added to an official release.  I am not a developer so I don't know that kind of stuff.  If you are going to use it I would recommend using something other than PID controller 0.  It flew awful until I switched to PID controller 1.  Now it is great and just needs some fine tuning.

http://www.rcgroups.com/forums/showthread.php?t=2249574&page=274&highlight=tricopter#post30946120

smarblewnec, 2015-03-05:
It shows that I closed the issue above my last post.  I did not mean to and must have clicked the wrong button.

nebbian, 2015-03-05:
Pull request here:
https://github.com/cleanflight/cleanflight/pull/591

thenickdude, 2015-05-07:
Merged now.

closed 2015-05-07 by thenickdude