cleanflight/cleanflight #550: Virtual Current meter calculations wrong?

pinkywafer, 2015-02-25:
I've found a few issues with the Virtual Current Meter:
I'm running Naze32 acro board.  LEDs on the frame (wired from PDB)
With Feature motor_stop, my armed AND disarmed current is the same - 0.6A
so I use current_meter_offset = 60
I calculated current_meter_scale = 166 (using the formula in the docs)

Problem 1: when DISARMED, telemetry and GUI is reporting current draw at 0.6A.  when I ARM, current is reported at 1.7A - but my draw is still 0.6A
   If motor_stop is disabled, this is not far off, BUT::
Problem 2: docs say to calculate using throttle min when ARMED, but if I have motor_stop disabled, and use current_meter_offset=170, then 1.7A draw is reported when disarmed, and this jumps again when armed.

I think the solution is:
1. change docs so that Imin (used for calculating both scale and offset) should be measured when disamed
2. override the current meter at minimum throttle if motor_stop is enabled

tracernz, 2015-02-25:
There is an error in the update I made to the docs last week sorry. I've corrected it in 872b1ea6955273eb98ddf0c9d26947d0f4796ac3 which I will submit a pull request for very soon. As for the current difference when armed and disarmed with MOTOR_STOP enabled, I'll look into this later but I thought that was accounted for already.

pinkywafer, 2015-02-25:
That looks good.
I've just been having a look at the code... motor_offset only seems to be taken into account in mixer.c when it's applied to the motor outputs, so I'm guessing it's not been applied to the rcCommand[THROTTLE], but I may be wrong
so where you have:
    `int32_t throttleOffset = (int32_t)rcCommand[THROTTLE] - 1000;`

just wondering if it needs:
   ```
int32_t throttleOffset = (int32_t)rcCommand[THROTTLE] - 1000;
if(rcCommand[THROTTLE] < rxConfig->mincheck){
    if (!feature(FEATURE_MOTOR_STOP))
        throttleOffset = escAndServoConfig->minthrottle - 1000;
    else
        thottleOffset = escAndServoConfig->mincommand - 1000;
 }
```
or something similar (I haven't spent much time looking at code recently!!)

pinkywafer, 2015-02-25:
Just realized this probably makes sense, I'm using Taranis and getting motor PWM reported on RPM on telemetry, and I think it's reporting 1150 when armed with motor_stop - I'm away from my quad at the moment, so can't check, if so, I'll raise another issue, but just checking the code, that is also getting its data from rcCommand[THROTTLE], so it may also need a fix.

tracernz, 2015-03-02:
I think if you change to
```c
int32_t throttleOffset = (int32_t)rcCommand[THROTTLE] - 1000;

if(rcCommand[THROTTLE] < rxConfig->mincheck){
    if (feature(FEATURE_MOTOR_STOP))
        throttleOffset = 0;
    else
        thottleOffset = escAndServoConfig->minthrottle - 1000;
 }
```

Unfortunately I don't currently have a toolchain setup so I can't compile anything at the moment and not a lot of free time to get it setup. Would you mind trying this out @pinkywafer ?

pinkywafer, 2015-03-02:
Ok, I'll have a look at it this evening

pinkywafer, 2015-03-02:
Just posted PR for similar issue... I think I may have to use ```escAndServoConfig->minthrottle``` instead of ```rxConfig->mincheck```  I'll get on it when I get home from work!

ledvinap, 2015-03-02:
IMO this won't help. Quadratic current approximation used in virtual current sensor will be usable only on much higher currents, so it make sense to use currentMeterOffset as disarmed consumption (quad may be disarmed but on for long period of time) and ignore low-current inaccuracy when armed.

If you want better precision, use more points for quadratic approximation and/or use motor command as source for current estimation.

pinkywafer, 2015-03-02:
I agree more points = more precision, but with motor_stop and motors not spinning (ie closed throttle) current draw is going to be the same as currentMeterOffset

ledvinap, 2015-03-02:
I checked my logs and current estimate is about half of actual value with minthrottle and FEATURE_MOTOR_STOP disabled. So it may actually gain some precision ...

Same check as https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L688 should be used..

hydra, 2015-03-09:
Ok, i think the fix in #589 is incorrect.

two points:
1) when 3D flight is used throttle off is midrc (~1500)
2) perhaps rc_controls/calculateThrottleStatus() should be used instead of directly accessing masterconfig which will drastically reduce the dependencies and avoid code duplication.
3) the PR causes the tests to fail.

since updateCurrentMeter() is called from annexCode() in mw.c the values that calculateThrottleStatus() expects could be passed in.

Can someone please update the PR to include the above changes and then I will merge.

pinkywafer, 2015-03-11:
I've updated #589 to use calculateThrottleStatus() as suggested and removed the now unnecessary dependencies, and this compiles fine and runs fine here, but I'm still getting failed Travis CI  - when RUNTESTS=True I'm getting undefined references to 'calculateThrottleStatus' and 'feature'.  I'd appreciate any help as I'm not sure how to fix this issue
Thanks

pinkywafer, 2015-03-16:
ignore above...being stupid! - I'd forgotten about unittests! - now done in PR

handsomejackuk, 2015-03-16:
just got this setup tonite and note also when disarmed with motor stop i am pulling 0.5a this is correct, but when armed and motors not running i am pulling 1.7a...  will keep on checking on this thanks for the great work....

hydra, 2015-03-30:
#589 is merged now, thanks everyone.

PR #589 (2015-03-30): bigfix virtual current meter
closes #550
closed 2015-03-30 by hydra