cleanflight/cleanflight #551: mixer.c L# 665 Code readability improvement

tricopterY, 2015-02-25:
Question for the experts:
```
maxMotor = motor[0];
    for (i = 1; i < motorCount; i++)
        if (motor[i] > maxMotor)
            maxMotor = motor[i];
    for (i = 0; i < motorCount; i++) {
        if (maxMotor > escAndServoConfig->maxthrottle)     // this is a way to still have good gyro corrections if at least one motor reaches its max.
            motor[i] -= maxMotor - escAndServoConfig->maxthrottle;
        if (feature(FEATURE_3D)) {
            if ((rcData[THROTTLE]) > rxConfig->midrc) {
                motor[i] = constrain(motor[i], flight3DConfig->deadband3d_high, escAndServoConfig->maxthrottle);
            } else {
                motor[i] = constrain(motor[i], escAndServoConfig->mincommand, flight3DConfig->deadband3d_low);
            }
        } else {
            motor[i] = constrain(motor[i], escAndServoConfig->minthrottle, escAndServoConfig->maxthrottle);
            if ((rcData[THROTTLE]) < rxConfig->mincheck) {
                if (!feature(FEATURE_MOTOR_STOP))
                    motor[i] = escAndServoConfig->minthrottle;
                else
                    motor[i] = escAndServoConfig->mincommand;
            }
        }
        if (!ARMING_FLAG(ARMED)) {
            motor[i] = motor_disarmed[i];
        }
    }
```
Any reason why it can't it be changed to:

```
maxMotor = motor[0];
    for (i = 1; i < motorCount; i++) {
        if (motor[i] > maxMotor)
            maxMotor = motor[i];
    }
    for (i = 0; i < motorCount; i++) {
        if (!ARMING_FLAG(ARMED)) {
            motor[i] = motor_disarmed[i];
        } else {
            if (maxMotor > escAndServoConfig->maxthrottle)     // this is a way to still have good gyro corrections if at least one motor reaches its max.
                motor[i] -= maxMotor - escAndServoConfig->maxthrottle;
            if (feature(FEATURE_3D)) {
                if ((rcData[THROTTLE]) > rxConfig->midrc) {
                    motor[i] = constrain(motor[i], flight3DConfig->deadband3d_high, escAndServoConfig->maxthrottle);
                } else {
                    motor[i] = constrain(motor[i], escAndServoConfig->mincommand, flight3DConfig->deadband3d_low);
                }
            } else {
                motor[i] = constrain(motor[i], escAndServoConfig->minthrottle, escAndServoConfig->maxthrottle);
                if ((rcData[THROTTLE]) < rxConfig->mincheck) {
                    if (!feature(FEATURE_MOTOR_STOP))
                        motor[i] = escAndServoConfig->minthrottle;
                    else
                        motor[i] = escAndServoConfig->mincommand;
                }
            }
        }
       /* if (!ARMING_FLAG(ARMED)) {
            motor[i] = motor_disarmed[i];
        } */
    }
```

thenickdude, 2015-02-25:
+1 for the if-else suggestion, though I would prefer to always put the positive branch first (the armed branch).

tricopterY, 2015-02-25:
I guess a quick check to perform useful work, won't hurt coding style vs spending time setting all sorts of values for nothing.

thenickdude, 2015-02-25:
Well, the extra CPU usage when disarmed is not really important, but the readability would be an improvement since it'd make it clear that none of the preceding code has any effect when not armed (so you don't have to carefully read through it looking for side-effects).

hydra, 2015-02-26:
yes, the inner loops should ideally cater for the normal case when armed.  yes to clarifying braces, etc.  submit a PR please.

hydra, 2015-02-26:
actually, don't bother, i will fix now.

hydra, 2015-02-26:
Done

closed 2015-02-26 by hydra

tricopterY, 2015-02-26:
Looks ten times better and more efficient. Thanks

commit 8282bc76 (2015-02-26): Mixer code readability and code size improvement.

Naze target size before: 113412, after: 113388

Closes #551