cleanflight/cleanflight #552: Too many PID controllers (difficult to choose best one)

matvey83, 2015-02-26:
Hi guys,

I'm new to Cleanflight, and one thing that I found confusing is that there are so many PID controllers, and each has its own caveats.   It's difficult to choose the right one, because before I make this decision, I  would need to tune each one, and then compare them.

IMO the user experience would be better if there was a single default PID controller that works relatively well most of the time (I suppose it's currently PID 0, but it apparently it has issues with PIDs affecting rates), and perhaps 1 or 2 "custom use" controllers, with a note explaining in what cases they are better or worse (and which ones support autotuning and self-level modes).

Thanks for considering!

thenickdude, 2015-02-26:
There's no need to try everything, PID controller 0 works just fine for
most situations.

hydra, 2015-02-26:
We have a few now yes. The plan is to delete the ones we don't want after people have been given time to find out which one works best.

closed 2015-02-26 by hydra