cleanflight/cleanflight #556: CC3D, HoTT Telemetry with main port or flex port did not work

CopterFail, 2015-02-26:
Using soft serial,the telemetry is fine. Using the physical ports, no response is generated by cf. Tested with: CC3D, clean flight 1.7.2 (and some earlier versions), MX12, GR-12S, protocol analyser.

tricopterY, 2015-02-26:
Is UART1/UART2 scenario set for telemetry?
eg.
feature TELEMETRY
set serial_port_2_scenario = 4
save

hydra, 2015-02-26:
Some points:

A) 1.8 will include updated serial/telemetry port handing
B)  this is likely just a mis-configuration issue - check your port configuration. There should be only ONE port with telemetry enabled on it in 1.7.2.
C) no details provided as to how you have it wired up - the FC only responds when it receives data for HoTT unlike with FrSky.
D) no configuration dumps provided.
E) there are incoming HoTT telemetry improvements.

Closing this now - if you still have issues after verifying your wiring and config are correct after 1.8 is released please re-open.

closed 2015-02-26 by hydra

CopterFail, 2015-02-26:
Is there a diffence in wiring / configuration to softserial? I've tested a 1N4148 and a simple 1k resistor, both work fine with the softserial, but fails with physical ports. 
The enhancement in #260 does not touch the port handling - in case you think about E). I've tested pierre's version too. 
The config is a full reset device with the suggested settings from serial.md (the port number are wrong there, since vcp is supported).