cleanflight/cleanflight #557: Sonar not working on Dragonfly Pro32

Sean295, 2015-02-26:
Running latest cleanflight and Naze32 firmware..activated Sonar, box is green at top, sonar sensor is at zero, stays there until i tilt the quad 45 degrees then it goes to -1, stays there until the quad is flat again then goes to zero. No matter hw quad is moved it only goes to -1. Using a spek satellite rx, trig and echo connected to M5 and M6, 5 volt and grd connected....tested on 2 sonar units..

hydra, 2015-02-26:
```
    // If we are using parallel PWM for our receiver, then use motor pins 5 and 6 for sonar, otherwise use rc pins 7 and 8
    if (feature(FEATURE_RX_PARALLEL_PWM)) {
        hcsr04_init(&sonarPWM56);
    } else {
        hcsr04_init(&sonarRC78);
```

Please connect using the correct pins, reopen if there's still an issue.  Remember to read the docs regarding the inline resistors.

closed 2015-02-26 by hydra

Sean295, 2015-02-26:
My mistake...didn't know when using serial rx to connect to the esc side. Added resistors...working great...thank you