cleanflight/cleanflight #558: Firmware wont load from On Line

clos8, 2015-02-26:
When I go to load firmware on line it says failed to load firmware. I have to download and load it locally. Any ideas? I am on the network.

hydra, 2015-02-26:
Yeah, something has changed with github/amazon s3, as of today from inside the Chrome debug console issue this:

`$.ajax({url:"https://github.com/cleanflight/cleanflight/releases/download/v1.7.1/cleanflight_ALIENWIIF1.hex"})`

results in:

` XMLHttpRequest cannot load https://s3.amazonaws.com/github-cloud/releases/9066347/6ee31b7a-ac62-11e4-8…TNZFOVBIJMK3TQ&Expires=1424972257&Signature=DgfKCgr05Q0byKmvisQlP98esYU%3D. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'chrome-extension://cfmnjnjafnagnjpojmbnikoeciedkkpl' is therefore not allowed access.`

Seems like a github/amazon S3 misconfiguration to me, setting up CORS headers on the preflight request's response is something github/amazon control.

hydra, 2015-02-26:
Fixed with this:

https://github.com/cleanflight/cleanflight-configurator/commit/400ac563c99389d61640b089ae0bf1f9f2cc43be

requires a release of the configurator :unamused:

closed 2015-02-26 by hydra