cleanflight/cleanflight #560: Some Harakiri PID controller updates and G-Tune support

MJ666, 2015-02-27:
I'm actually working on G-Tune support for CF and a few minor Harakiri PID controller updates. Here is my development repository:

https://github.com/MJ666/cleanflight/commits/Harakiri_PID_update

The coding is completed and I will go into testing phase now. I can do some initial tests but for an wider test I need some interested and experienced pilots to help.

Here is the related documentation:

https://github.com/MJ666/cleanflight/blob/Harakiri_PID_update/docs/Gtune.md

The formerly hardcoded parameters for the Harakiri PID controller are now also configurable.

set pid5_maincuthz = 12     [1-50Hz] Cuf Off Frequency for D term of main Pid controller
set pid5_oldyw = 0               [0/1] 0 = multiwii 2.3 yaw (default), 1 = older yaw

Some feedback is appreciated. Thanks.

vadav, 2015-02-27:
Cant wait to try this out  =)

strips, 2015-02-27:
I can test G-Tune if you got a binary for Naze.

MJ666, 2015-02-28:
I have finished the initial tests and have posted an Hex fro the Naze here:

http://www.rcgroups.com/forums/showpost.php?p=30919237&postcount=3897

Please read this post before you go ahead. I also can provide test binaries fro other targets if there is interest.

MJ666, 2015-02-28:
Did some more flight testing at the flying field. Please see here:

http://www.rcgroups.com/forums/showpost.php?p=30922850&postcount=3920

vadav, 2015-02-28:
Tested the hex you provided for the naze, and found front/right drift in acro mode. Tried to use gtune, but nothing was saved...  activated gtune while in air, flew around and made sure not to touch stick for as long as i could..  with some throttle bumps..   landed, disarmed and turned off gtune..  as your doc say "the tuning results will only be saved if you disable G-Tune mode while the copter is disarmed", but nothing saved..

MJ666, 2015-02-28:
I have tested this a couple of times (10-15 flights) and it was working for me. Was also testing on PID 0, PID 3 and PID 5. I have 3 profiles configured. On the NAZE you can se when it is saving. One of the LED's is flashing during save. I usually activate G-Tune already before takeoff but it should get active any time. Tuning will happen anytime when the sticks are released and there is no input. this is also independent on the axis. pitch tuning will be done wen there is no pitch input. There could be input on the other axis. You fly different profiles? Do you have G-Tune AUX configured in all of them? I was missing this initially and was wondering there was no reaction sometime.

vadav, 2015-02-28:
My test was in pid controller 5 with only one profile in use. 

Can you confirm that you have to disarm before turning Gtune off, to make it save results.

MJ666, 2015-02-28:
Yes first disarm and the switch G-Tune of. Switching G-Tune off before disarming will not save but will keep the new P values in memory as long you will not power off.

BTW: I use a switch for arming. Not sure if this makes a difference? Normally it should not.

vadav, 2015-02-28:
All ok now, stupid me didnt have trims centered..    all working as it should!!   =)

MJ666, 2015-02-28:
This was likely also your drift :-)

vadav, 2015-02-28:
Sure was..   now im praying the weather gods are nice tomorrow..    =)

vadav, 2015-02-28:
While we're at it..  what does this do?   set pid5_maincuthz = 12 [1-50Hz] Cuf Off Frequency for D term of main Pid controller

MJ666, 2015-02-28:
Please see here:
http://www.rcgroups.com/forums/showpost.php?p=30910398&postcount=3819

strips, 2015-03-01:
I had a 30 minute window to play inside today :) 

I tested on PIDC 2 and flicking on G-Tune upped all Ps to 20 in GUI!!! It didn't have enoug power to lift off. Just wobbling on the floor. When I restored old PIDs and turned it on 50cm above the flor it started shaking violently and tumbled to the floor. I did not have more time to try limiting the max P. But it seems like G-Tune starts out at max?

Too see if I got similar effect on PIDC5 I really do not know if anything happened at all or I switched it off before disarming as I did not see any PID changes. It flew with no apparent changes. My wife and kids got home at this point and no more testing today :(

PS. That wobble almost burned my motors. They got pretty warm and started to smell. I hope I didn't damage them. I have heard the KISS (w/Oneshot) can quickly overheat motors if you got oscillations.

MJ666, 2015-03-01:
The default max for P is 7.0. Did you modified the default settings? The range for P is between 2.0 and 7.0 (default, but can be changed in the CLI). If the starting P is in-between this range it will start with this one if it is above the range it will start with the configured minimum. If you fly in Level or Horizon the roll and pitch PID's should not be touched. What was your starting P if you get 20 with the default settings this is a bit strange and I can not explain. Just was checking the code again since PID 2 is using float and I added the special handling for this. Will see if I can test pid2 on one of my Nano's.

MJ666, 2015-03-01:
Found the problem for LuxFloat and G-Tune. committed a fix an will publish new HEX here:

http://www.rcgroups.com/forums/showpost.php?p=30933507&postcount=4026

strips, 2015-03-02:
Did some quick flights in the dark now. Looks much better. I had only one charged battery so I just managed to quickly test PIDC 2 and 5. I have a slight yaw drift and I presume thats why I did not see any change in yaw P.

At least it behaved perfectly and should not be a problem to implement in main release. At least when thinking of the other autotune still in the main release.

Posted som pics of PIDs here http://www.rcgroups.com/forums/showpost.php?p=30941917&postcount=4078

I tested this on a stiffened Nemesis 240 frame. 3S 2200mAh, Tiger 1806 2300kv, HQ 5x4.5 bullnose, AUW is 750g or was it 800g :o I got a lot of shit on it ;)

acceleroto, 2015-04-11:
I flew a G-Tune flight today and it seemed to work fine.  I built new firmwares from a clone of the MJ666 Harakiri_PID_update branch last night.  Here are the Naze32 and Sparky hex files I built from that branch (I flew that Naze build, but not the Sparky build) -
http://rotorland.com/share/cleanflight/gtune_test/cleanflight_gtune_v1.9.zip  (Back up your configuration first -- loading that firmware wiped all my settings.

I flew in pid_controller 5 with a setup that was fairly well tuned for my quadcopter.  If anything, my copter was a little sluggish in pitch, but roll felt solid.  Before I flew, I set my TPA to 0 (it was 0.3).  I took off in angle, then switched to acro and turned on G-Tune.  I flew for about 3 minutes trying to keep my hands off the sticks as much as possible.  I landed in acro mode, then disarmed.  I connected to Configurator and took a screenshot of the PIDs.

I must have misunderstood how to make save the PIDs by flipping switches, because what I tried didn't work.  I landed in acro & G-Tune, then disarmed, flipped to G-Tune, then back off...then connected to Configurator.  The lights on the Naze did not show that it was saving, so at least that matches.  Several other people I spoke with on #cleanflight were also confused at the correct procedure to save values.  Can someone post the exact step-by-step instructions for how to make G-Tune save the new PIDs?

The G-Tune PIDs are here: http://imgur.com/QkMnwu0  The old P values were Roll: 3.5, Pitch: 5.0, Yaw: 8.4.  The new values were 3.4, 7.0, 7.0.  All other values can be seen on the image above (they're unchanged since G-Tune doesn't touch them).  Of note, I didn't change the max values for G-Tune, so 7.0 matches the max value.  I only did one run, so it'd be good to do more for comparison.

The new PIDs flew well, except for some slight oscillations in roll at high power settings.  I'll add some TPA back in to make it go away.  I may also bump up the max allowed P in pitch and yaw to see what happens there.

Setup:  ZMR250, Naze32 full, KISS v1.1 18A ESCs in Oneshot, Emax MT1806-2280 motors, Gemfan 5030 props, FPV gear & Mobius.

thenickdude, 2015-04-11:
@MJ666 I tried adding support for decoding the G-Tune blackbox events, but I noticed that you're logging the gtuneGyroAvg as a single unsigned byte (with `blackboxWrite()`). This value probably should be stored as int32_t in the struct and written with `blackboxWriteSignedVB()` instead so the gyro reading doesn't get truncated.

MJ666, 2015-04-12:
Make sense. I will make an related commit. Have found some other issues related to the BlackBox code and will fix them.

acceleroto, 2015-04-12:
All-  I removed the Sparky build from my zip file above since Sparky isn't supported yet.

@MJ666 Thanks for the step-by-step.  I'll try that.  I landed with G-Tune on the other day and dearmed with it still on.  Oops.  I'll also bump up my deadbands.  It's almost always windy here, so I'll hunt for a calm morning to test.

MJ666, 2015-04-13:
I have added Sparky support to the PR an published an build on RCG forum.

acceleroto, 2015-05-11:
I flew G-Tune again today, but this time with new motors on my 250.  I followed MJ666's procedure above on the correct switch order.  G-Tune worked as it should and saved the PIDs as expected.  I flew again after the G-Tune flight to check the resulting P values.  As before (and as expected), the P values were slightly too high.  At the end of full stick rolls, there are several overshoots (wobbles) when I center the stick.  They're not gross, but it's enough that they're easily noticeable in videos.  I'm guessing I need to back off the P values at least 10% -- it may be more though.  The weather got bad before I could fly a third time.  It seems like either some logic tweaks in determining P values should be added or maybe a final multiplication fudge factor when the P's are saved.

hydra, 2015-10-06:
I want to merge this in 1.11.0.

hydra, 2015-10-06:
merged.

closed 2015-10-06 by hydra