cleanflight/cleanflight #563: SSG (Super Simple Gimbal)

danielolsman, 2015-02-27:
Is there a possibility to add the SSG code

more info is found here : http://www.rcgroups.com/forums/showthread.php?t=1793759
Multiwii code is found here : http://www.multiwii.com/forum/viewtopic.php?f=8&t=3079

With this code you could use 2 servo's as a simple gimbal

SlyMike, 2015-02-27:
Its already supported!
 Tick the "SERVO_TILT" check box in the configurator app

danielolsman, 2015-02-27:
i don't think SSG is supported.

This is what i mean:

https://www.youtube.com/watch?v=KPBL4pPEzNI\

SlyMike, 2015-02-27:
Yeah, I watched that. It's actually quite an innovative design, maybe it will end up on someone's to do list.

olexs, 2015-02-27:
It's supported. See https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md and specifically the "gimbal_mix" parameter - you want "set gimbal_mix=2".

SlyMike, 2015-02-27:
@danielolsman looks like you can close this issue thanks to olexs keen eye.

danielolsman, 2015-02-28:
Ok, thanks for the comments. I will try it. 
issue closed.

closed 2015-02-28 by danielolsman