cleanflight/cleanflight #566: Can we standardize line lengths?

timkostka, 2015-02-28:
Just to get the discussion going...

Can we standardize the line lengths within cleanflight?  The MultiWii style guide doesn't really mention this at all.  I find it quite hard to read sections of the code that have 200+ character lengths with all the horizontal scrolling required.  It's even harder if you want to put two windows side-by-side.  I've used an 80-character rule in the past and it has worked quite well for readability.  It's also what Google (and a bunch of others) recommend in their style guide: http://google-styleguide.googlecode.com/svn/trunk/cppguide.html#Line_Length

I'm a fan of the following:
* New code should generally adhere to 80-character limit per line.  Exceptional cases where readability would suffer significantly can be excluded.  Old code does not need to be immediately ported.

I did the following example to show the difference.  This is a section of the current code (taken at random):

```
void updateInflightCalibrationState(void)
{
    if (AccInflightCalibrationArmed && ARMING_FLAG(ARMED) && rcData[THROTTLE] > masterConfig.rxConfig.mincheck && !IS_RC_MODE_ACTIVE(BOXARM)) {   // Copter is airborne and you are turning it off via boxarm : start measurement
        InflightcalibratingA = 50;
        AccInflightCalibrationArmed = false;
    }
    if (IS_RC_MODE_ACTIVE(BOXCALIB)) {      // Use the Calib Option to activate : Calib = TRUE Meausrement started, Land and Calib = 0 measurement stored
        if (!AccInflightCalibrationActive && !AccInflightCalibrationMeasurementDone)
            InflightcalibratingA = 50;
        AccInflightCalibrationActive = true;
    } else if (AccInflightCalibrationMeasurementDone && !ARMING_FLAG(ARMED)) {
        AccInflightCalibrationMeasurementDone = false;
        AccInflightCalibrationSavetoEEProm = true;
    }
}
```

With an 80-character line length, it becomes the following instead:

```
void updateInflightCalibrationState(void)
{
    // Copter is airborne and you are turning it off via boxarm: start measurement
    if (AccInflightCalibrationArmed &&
            ARMING_FLAG(ARMED) &&
            rcData[THROTTLE] > masterConfig.rxConfig.mincheck &&
            !IS_RC_MODE_ACTIVE(BOXARM)) {
        InflightcalibratingA = 50;
        AccInflightCalibrationArmed = false;
    }
    // Use the Calib Option to activate:
    //   Calib = TRUE Meausrement started, Land
    //   Calib = 0 measurement stored
    if (IS_RC_MODE_ACTIVE(BOXCALIB)) {
        if (!AccInflightCalibrationActive &&
                !AccInflightCalibrationMeasurementDone)
            InflightcalibratingA = 50;
        AccInflightCalibrationActive = true;
    } else if (AccInflightCalibrationMeasurementDone && !ARMING_FLAG(ARMED)) {
        AccInflightCalibrationMeasurementDone = false;
        AccInflightCalibrationSavetoEEProm = true;
    }
}
```

How does everyone else feel about it?

tricopterY, 2015-02-28:
+1 Definitely much easier to read and comprehend. I hope any additions/updates to the code, could add more inline comments, to help newcomers like myself to follow the code and be able to contribute.
After all, this is supposed to be an open collaboration.

ledvinap, 2015-02-28:
speaking of style: 
```
    if (AccInflightCalibrationArmed
        && ARMING_FLAG(ARMED)
        && rcData[THROTTLE] > masterConfig.rxConfig.mincheck
        && !IS_RC_MODE_ACTIVE(BOXARM)) {
```
is more readable, the operator is before item it binds to ..

tricopterY, 2015-02-28:
I guess not all functions in CF need a description, as the naming convention is pretty good. 
But, any new functions should provide the basic function description. May be someone would like to document the API in future.
/**
* @brief This function updates inflight calibration state and flags it to be saved to eeprom.
*  Only called if FEATURE_INFLIGHT_ACC_CAL is  set.     
* @param None
* @retval None
*/
void updateInflightCalibrationState(void) 
{
}

tracernz, 2015-03-01:
I agree on the line length. I also agree that functions should have doxygen style descriptors. I'm not so sure about adding too many inline comments, what you end up with there is comment rot and then they're the opposite of helpful. Well written code should read almost like English anyhow.

hydra, 2015-03-02:
code should document itself where, ever possible.  comments rot and can't be refactored automatically.
i don't want to be draconian with line lenghts.

two simple rules apply:
1) do less per line - makes diffs much easier to read and the lines never get long in the first place.
2) if you have to scroll sideways to read it, it's too long.

hydra, 2015-03-02:
A PR with some coding/style guidelines might be useful now.

closed 2015-03-02 by hydra