cleanflight/cleanflight #569: [issue] Strange barometer altitude spikes

digitalentity, 2015-02-28:
I have a BMP180 sensor connected to my CC3D and I noticed that it produces strange spikes every few seconds. This is likely not a hardware issue as I have connected a BMP085 breakout board and seen these spikes as well. Configurator does not show any I2C errors.
[image]

SlyMike, 2015-03-01:
Do you have an i2c oled display connected? I noticed similar behaviour with my nase32 the other day. I was getting an occasional spike but also noticed a stepping pattern imerge as the pages changed on the oled display.

thenickdude, 2015-03-01:
Hm if that's the case then maybe the barometer code doesn't tolerate
variance in the looptime (I'm assuming that updating the display creates a
lag spike).

SlyMike, 2015-03-01:
Just tried to recreate my trace for you and ended up with this instead...

[image]

Not what I expected! 
Currently running Cleanflight/NAZE 1.7.2 Feb 23 2015 / 14:21:30 (afe44d1). I got the stepping pattern on 1.7.0.

SlyMike, 2015-03-01:
Flashed back to 1.7.1 to see if the original steps were there...

[image]

[image]

[image]

[image]

^^^ that's what I'm talking about ^^^

[image]

[image]

^^^ Back to normal ^^^

@digitalentity Sorry for hijacking your thread!

digitalentity, 2015-03-01:
@SlyMike, no problem, we seem to have a common problem here. 

I've noticed alot of spikes when motors are running at high throttle, this is probably a noise in powerlines issue. Will to some more testing in a few days.
Also ordered a new baro, this time a MS5611, to see if the issue is indeed hardware related.

thenickdude, 2015-03-02:
WIth props attached? Because with props on, you expect spikes at high
throttle due to prop wash.​

digitalentity, 2015-03-02:
@thenickdude Spikes due to prop wash are expected and present, but they are not a 25 meters variance. Moreover, I have my baro covered by piece of HEPA to prevent high speed air flow from messing with baro readings.

hydra, 2015-03-05:
I confirm that when disarmed you will get baro spikes when an OLED display is connected, especially when changing pages.

The baro code is coupled to the looptime, it needs decoupling.

This isn't an issue when armed since the display does not update.

hydra, 2015-03-05:
Regardless of OLED display, this still needs fixing.

digitalentity, 2015-03-05:
I don't have an OLED connected, just usb vcp, mag, baro and gps.

thenickdude, 2015-03-16:
I couldn't reproduce this on my Naze32 rev 5 (which I believe is the MS5611 barometer), even when I added [this code to mw.c](https://github.com/sherlockflight/cleanflight-dev/commit/664c932f008042f455bc1e0c5f82e056ad343885) to add crazy high random variability to my looptime (i.e. add between 150ms and 0ms delay to looptime):

[image]

I do see a 0.5 - 1 meter spike precisely as baro calibration finishes (which takes a long damn time at this looptime). This one is 0.5 meters:

[image]

I can cause the altitude to vary by +-0.5 meters just by holding my hand over the top of the board.

I notice that the BMP085 detects the situation where you try to read the value from the baro before it's ready, and does nothing with that information (it just tries to read from the device anyway):

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/barometer_bmp085.c#L301

That would be my pick for a BMP085-specific spike generator.

digitalentity, 2015-04-01:
Just tested a newly received BMP180 breakout board, same random spikes, even when disarmed. I was unable to reproduce these spikes on Flip32 with MS5611 board, so this issue is unrelated to looptime. It is also not related to a specific sample of BMP085/BMP180 sensor as I am seeing these spikes on 3 different boards. I think this is likely software-related. I will look into the code and see if I can come up with a fix for this.

ledvinap, 2015-04-01:
I have probably same problem on sparky. USB usage is probable cause here, looptime goes to ~20ms range ...

ledvinap, 2015-04-01:
@thenickdude : The problem is probably not directly related to looptime difference, but to stalling of whole main loop. Adding random delay_ms() to main loop() may simulate it ...

thenickdude, 2015-04-01:
@ledvinap That's what I did test though:

https://github.com/sherlockflight/cleanflight-dev/commit/664c932f008042f455bc1e0c5f82e056ad343885

Causes the main loop to have a random very large delay at the beginning of it.

ledvinap, 2015-04-01:
Yes ... 
 `delay(301 * (currentTime % 511));` should do the trick ...

hydra, 2015-04-01:
@thenickdude yup, that's the kind of thing I see.

digitalentity, 2015-04-01:
I've just tested the same - added the random delay and got very similar picure. Seems this is not related to baro driver after all.

hydra, 2015-04-01:
@digitalentity yes, that's what I thought when I commented on the issue a month a go. Good to have confirmation.

digitalentity, 2015-04-01:
It seems the root of the problem is somewhere in calculateEstimatedAltitude(), I'll further investigate this tomorrow if I have some time to spare.

digitalentity, 2015-04-01:
Did a quick hack to bmp085 driver to fix preassure to a predefined value and introduce some random variance (19ab9261c0315115826320dce6628703a2ac3ef3) and was very surprised - the spikes were all gone! This is a bmp085-specific problem after all.

digitalentity, 2015-04-01:
Looks like bmp085/bmp180 does not always follow the datasheet when it comes to conversion delays. I've increased ut_delay and up_delay to much higher values (da9fdd47842571c2611f52c2b4a4666388e9b60e) and spikes are not present any more - just normal altitude fluctuations. It is a hack, but it seem to do the trick for me.

thenickdude, 2015-04-02:
It's not bmp085 specific as the timing variation I saw was on the Naze32, which uses MS5611. They could both have the same bug in their drivers though. Also adding delays to the loop shouldn't ever cause the barometer to be read faster, so it should be eliminating that if it was the cause.

I'm confused by what you're trying with adding random values to the read pressure?

digitalentity, 2015-04-02:
@thenickdude by adding random variance I was trying to simulate preassure changes. If I don't add random variance altiude is zero and completely constant without any spikes or changes regardless of looptime.
Then I altered the driver itself and increased the timeout between 'read pressure' command and actual reading raw value from the sensor. Spikes I was seing in my first comment are gone. I think that bmp085 hardware does not always behave exactly by datasheet, sometimes it need more time than 25.5ms (as in datasheet) to finish the pressure conversion.

thenickdude, 2015-04-02:
Those spikes are probably due to the bug in the driver that tries to read the pressure/temperature value anyway even though the code knows they aren't available yet (by waiting for the flag to be set). That check can probably be fixed without increasing the delay:

```
    // wait in case of cockup
    if (!convDone)
        convOverrun++;

    i2cRead(BMP085_I2C_ADDR, BMP085_ADC_OUT_MSB_REG, 2, data);
    bmp085_ut = (data[0] << 8) | data[1];
```

(notice how when convDone is false it still reads it anyway and returns the value!)

digitalentity, 2015-04-02:
convDone is used only if we have an additional wire from EOC pin on BMP085 to some EXTI line. External sensors on I2C does not have that, so this code should not be even compiled when BARO_EOC_GPIO is undefined.

thenickdude, 2015-04-02:
Ah, good point.

digitalentity, 2015-04-02:
I've cleaned up the code and removed convDone when BARO_EOC_GPIO is not defined (f9c1b63959479f8dfac440fe90709559a00693ca). If this worth merging, I can create a PR.

digitalentity, 2015-04-02:
Had a two hour bench test with this code -  works good, no spikes in altitude.

ledvinap, 2015-04-02:
If the spikes are caused by reading occasionally register too fast, using short median filter should greatly improve the situation.

digitalentity, 2015-04-02:
Barometer code already uses averaging, doesn't help much. We we can simply
read our bmp085 baro less frequently (increase a timeout) and this will
eliminate the problem. I left my flight controller running for 12 hours
with increased baro timeout - no spikes in altitude were recorded.

ledvinap, 2015-04-02:
@digitalentity : Averaging filter won't help when the spike is orders of magnitude larger than normal values. Median filter will simply discard outliers, with minimal effect on valid values.

digitalentity, 2015-04-02:
Sorry, had to refresh my memory on what a median filter is. I guess we'll need to decide what is better - reading bmp085 baro every 33ms with occasional errors and spend CPU time filtering them out, or every 50ms without errors.

ledvinap, 2015-04-03:
Doing both is probably best option. With current filtering and altitude-hold mode, single spike will probably crash quad  ...

digitalentity, 2015-04-03:
Ok, I'll open a PR for fix in bmp085 driver and will work on median filtering for generic barometer sensor (barometer.c), or probably the best place to apply median filtering is altitude.c where altitude is calculated from several sources.

ledvinap, 2015-04-03:
The problem is that median filter won't work well after any low-pass filter. It would be probably best to filter ADC readings, but it is hidden inside driver code, limiting code reuse ...

digitalentity, 2015-04-03:
@ledvinap I've added a generic int32_t implementation of 3- to 9-point median filter and modified generic barometer code to use a 3-point version, should be independent from actual barometer hardware driver. Please see commit 963e8f37b3678759cd25b140a2dd02eaa286d0bf. Median filter code can also be reused for sonar etc.
Didn't test the code though, don't have a cc3d board at home.

ledvinap, 2015-04-03:
Well .. the baro problem is probably caused by wrong timing:
- currentTime is set in `loop()` 
 - https://github.com/cleanflight/cleanflight/blob/7efdcf13e340169ba57d86142396afd4010411f5/src/main/mw.c#L669
 - https://github.com/cleanflight/cleanflight/blob/7efdcf13e340169ba57d86142396afd4010411f5/src/main/mw.c#L676

but this variable is used in `executePeriodicTasks()` for `baroUpdate(currentTime);` (on next loop and possibly skipping once because of `shouldProcessRx()`. The delay between sampling time and executing `baroUpdate` may be quite long and affected by other tasks. If this delay is long when start-conversion command is send and short when reading data, baro value won't be updated ... 

Correct action is probably to call micros() again after `switch(state)`, so that wait is guarantied to be longer than requested.

ledvinap, 2015-04-03:
Fix in https://github.com/cleanflight/cleanflight/pull/720, fixes problem on sparky.

Other functions called with `currentTime` should be checked too ..

digitalentity, 2015-04-03:
Your fix is better than mine, it eliminates the cause. Modified #718 to solve end-of-conversion flag not being used when available.

digitalentity, 2015-04-03:
Other functions called with 'currentTime' and affected are probably these:
https://github.com/cleanflight/cleanflight/blob/1e7fb08b9d8afc98945b0eb77b6b55ce55c43888/src/main/rx/rx.c#L313
https://github.com/cleanflight/cleanflight/blob/2a73ad69253be498611b79b38c8f6bb52990f774/src/main/sensors/compass.c#L72
https://github.com/cleanflight/cleanflight/blob/2a73ad69253be498611b79b38c8f6bb52990f774/src/main/io/statusindicator.c#L53
https://github.com/cleanflight/cleanflight/blob/4746b336b9870bf98f0df237438ec4db9bbb7c69/src/main/io/gps.c#L1065

ledvinap, 2015-04-03:
Only compass is possibly affected and both compass chips now in use should simply return previous value/wait for ready bit.

digitalentity, 2015-04-03:
Than the only piece of code that should be changed is baro-related.

ledvinap, 2015-04-03:
It is probably enough to fix problem with baro now, and maybe fixing compass code to avoid surprise in future.

digitalentity, 2015-04-03:
I've tested the 3-point median filter on buggy bmp085 code, spikes are all gone. This does not fix the cause of spikes (bad timing) but effetively filters them out.

digitalentity, 2015-04-03:
@ledvinap as a test, I've merged your's #720 and mine #718 and #725 into current master and bench-tested the code. All seem to work fine - baro provides sane readings with no spikes.

ledvinap, 2015-04-03:
IMO the median filter is good idea even with fixed code - single failed I2C read may cause big problems otherwise ...

digitalentity, 2015-04-03:
I think the same way. Probably the same median filter should be applied to sonar readings as well.

digitalentity, 2015-04-04:
Did a flight test today. Baro mode works really well now. Quad was hovering in BARO + GPSHOLD mode for about 10 minutes and I didn't notice any sudden altitude changes, everything was within normal 0.5-1 meter variance.

hydra, 2015-04-07:
[image]

So, after merging the the PR's #720, #718 and #725 I observe the altitude calculation is still incorrect when an OLED display is connected when disarmed.

Granted, an OLED display doesn't change pages or refresh when armed however, if i'm understanding things correctly looptime fluctuations cased by other long running tasks that happen when armed will still cause issues, e.g. processing a new GPS frame.

What other fixes can be made?

hydra, 2015-04-07:
Also, thanks for the great investigation work, fixes and testing so far!

digitalentity, 2015-04-07:
Don't have an OLED display to investigate further. When I receive my brand new display from China, I'll dig into this issue again. So far I am very happy with the results I am getting.

hydra, 2015-04-08:
@digitalentity You can replicate the behaviour it by adding a variable delay in exectutePeriodicTasks() that changes every 5 seconds and is executed at the same frequency that the display would normally execute (5hz).

hydra, 2015-04-08:
@ledvinap @digitalentity I'm thinking that perhaps the call to `executePeriodicTasks()` should be performed on every loop iteration, even iterations that would call `processRx()`. Comments?

hydra, 2015-04-08:
Perhaps this commit would help too: eb1617c6d88a92db3870bd8561c09b862b0d0d05

ledvinap, 2015-04-08:
@hydra: You are talking about 1m fluctuation?
Try disabling all attitude filters. The baro noise about 10m on my Sparky, the problem you see may be some aliasing problem. Or voltage fluctuation. Or black magic ...

ledvinap, 2015-04-08:
What about implementing priority queue for timed task? Just register periodic invocation or invocation delay. Then just fetch first callback in `loop()` and call it. 

With some care it may handle CPU overload much better that current scheme.
It may be also possible to include 'expected runtime' in callback descriptions and reorder tasks to execute high-priority tasks with better accuracy (Well ... much work was probably done in this field in eighties ...).

This should work without introducing much thread synchronization problems that would come with full RTOS. And maybe it could be slowly migrated to cooperative multitasking (GPS code is great candidate) ...

hydra, 2015-04-08:
@ledvinap yes, ~1m fluctuation on the SPRacingF3 with OLED enabled.  ~10m fluctuation on Sparky with all settings at default.

While I agree that better task scheduling is required at some point, is there nothing else we could do now instead?

ledvinap, 2015-04-08:
The 10m on sparky is after default filtering? Can't test right now, but that would be terrible.

Using better filtering may help a bit, but noise should be reduced as `sqrt(averaging)`, so overaging 100 samples may result in ~1m fluctuation.

Randomizing baro sampling times may help if main problem is caused by aliasing.

Did you try to disable all filters and use blackbox to record baro data? Analyzing the data may reveal something interesting ...

hydra, 2015-04-11:
@ledvinap My bad - On the sparky i get about 3m fluctuations using all default settings.

[1 more reply]

closed 2017-02-25 by hydra