cleanflight/cleanflight #572: Ledstrip indication issue

Fvk1960, 2015-03-01:
I have 2 ledstrips with each 8 leds.
I want to configure led 0,7,8,15 as corner indicator and remaining as ring.
if i wire whole 16 led at top line of the grid the indicator leds for pitch and roll seem to determine themselves there orientation regarding NSEW. Led1 assumes W all other assume E. 
Even if i assign a compass direction to these leds this is not taken over.
My target is 
LED0+7 Roll left, LED8+15 Roll right.
LED0+15 Pitch backward
LED7+8 pitch forward.
I could only achieve this by putting LED0 and 15 one line loewer in the grid without assigning orientation to these leds only mark as indicator.

Second issue at full pitch forward the leds 7+8 did not blink or but stop blinking after a few seconds.
When i reduced the range on my transmitter with 1% it was ok (150%  to 149% devo7 actual ouput is around 2000.

Some request the parameter INDICATOR_DEADBAND can this be made configurable via cli ?

hydra, 2015-03-02:
please:
a) provide configuration dump of the configuration you tried that didn't work for you
b) provide a screenshot of the led tab
c) raise one issue per ticket

bhuism, 2015-03-04:
I have the same, at first the leds are fine, and react as programmed, after some led conf changes, they become erratic and don't respond as programmed. I'll post a new issue with a) + b) later

hydra, 2015-03-05:
@bhuism your problem is probably due to your LED VCC and DATA voltages - check documentation.

@Fvk1960 need steps to reproduce and your config dump otherwise this cannot be investigated.

Fvk1960, 2015-03-13:
for @bhuism at first my ledstrip did not work until as documented a added a diode in the power lead to he ledstrip.
I will make screendumps later

bhuism, 2015-03-13:
Hai, got it sorted, I did not read the docs on this point, with the diode's all is fine, thanks!

Fvk1960, 2015-03-15:
[image]
Ok this setup Roll left gives led0 and led1,Rol right gives led 14 and led 15
Pitch forward [2000] gives all 4 leds
HOWEVER Pitch backward [1000] no leds are lid
[image]
Assign orientaion has no effect.
[image]
Relocating led0 and led15 one row lower on the grid works ok
Now pitch forward led1 and led14 lids
And pitch backward led0 led15 lids
working dump
# version
Cleanflight/NAZE 1.7.2 Feb 23 2015 / 14:21:30 (afe44d1)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature SONAR
feature LED_STRIP

# map
map AETR1234

# led
led 0 0,1::I:0
led 1 1,0::I:2
led 2 2,0::RC:6
led 3 3,0::RC:6
led 4 4,0::RC:6
led 5 5,0::RC:6
led 6 6,0::RC:6
led 7 7,0::RC:6
led 8 8,0::RC:6
led 9 9,0::RC:6
led 10 10,0::RC:6
led 11 11,0::RC:6
led 12 12,0::RC:6
led 13 13,0::RC:6
led 14 14,0::I:2
led 15 15,1::I:0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
[... 555 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 165
set i_vel = 125
set d_vel = 5

# dump rates

# rateprofile
rateprofile 1

set rc_rate = 100
set rc_expo = 0
set thr_mid = 60
set thr_expo = 70
set roll_pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

#

Fvk1960, 2015-03-28:
Why blocked status with no feedback.
Ok i will close this item.

closed 2015-03-28 by Fvk1960