cleanflight/cleanflight #575: Feature request: rotation limit

agrath, 2015-03-02:
Suggestion for when in "rate" mode, aka not horizon/angle
A limit to maximum rotation off level angle for pitch/roll; the copter would not return to level but you could not rotate it past more than 45 degrees (configurable)
You would still have manual control over the angle off level, just like normal rate, but you wouldn't be able to over-rotate the quad.
This would be great for people trying to learn how to fly in rate mode when they're used to an auto-level flight controller.
This might be possible to do already by using horizon with a extremely low "I gain" so it never actually returns to level

magnazan, 2015-03-02:
This is exactly the same as what you can do with the Eagletree Vector mode called 2d. Very handy and great for just cruising around.

+1

vadav, 2015-03-03:
Best way to achieve this would be the ability to turn on/off the leveling in Angle mode, as we already have a max angle setting in cli for that..

dkisselev, 2015-03-10:
I feel like semantically speaking, it would make more sense to recreate the feature in rate mode.

Angle is known to be a self-leveling mode, so users expect that out of it. It's also documented as 'legacy', so it wouldn't make sense for new features to be developed against it.

Just need to implement the max inclination code in the rate flight code and add a parameter to disable/enable it.

hydra, 2015-05-15:
With reference to the title changes i got confused :smile:

closed 2017-02-25 by hydra