cleanflight/cleanflight #578: Failsafe causes props to spin even though throttle was at zero

sirbadger, 2015-03-02:
Board was armed and throttle at 1007, signal lost to RX, RX set throttle to 920 (failsafe on RX), RX regained signal, board span motors up briefly to 1200, settings:

failsafe_delay = 10
failsafe_off_delay = 200
failsafe_throttle = 1200
failsafe_min_usec = 985
failsafe_max_usec = 2115

Should the failsafe kick in and spin up motors even if throttle was at zero?

hydra, 2015-03-03:
I'm failing to see how the failsafe kicked in if your RX was sending a signal all the time.

the failsafe _only_ kicks in if no signal is received from the RX.

see:

https://github.com/stone/cleanflight/blob/master/src/main/rx/rx.c#L207-L211

and here:
https://github.com/stone/cleanflight/blob/master/src/main/rx/rx.c#L257-L262
https://github.com/stone/cleanflight/blob/master/src/main/rx/rx.c#L276-L278

Please provide exact steps to repeat.

How do you know the motors were span up to 1200?  1200 is the throttle input to the pid controller, not the motors.

what receiver are you using?

how long was the signal lost for before RX regained signal?

sirbadger, 2015-03-03:
The RX (OrangeRX R615X) drops the throttle to 920 when its failsafe kicks in.

Whilst watching the receiver tab it very briefly jumped to 1200 after signal was restored.  Perhaps this is a coincidence but after setting failsafe_throttle to 1000 it jumps to this value after signal is restored but obviously the motors do not spin.

The receiver was off for only a few seconds.  I'm not sure if its related but once failsafe has happened and reconnected the cycle time jumps to ~ 47000 when the board is armed, when you disarm it drops back to 2500.

I'll provide further details of steps this evening (UK time)

sirbadger, 2015-03-03:
For clarify i should have said the throttle rc input hit 1200.  I cant be sure of the motor output.

sirbadger, 2015-03-03:
Ive just tried again with failsafe_throttle = 1300, turned transmitter off, RX set throttle to 920 (this is the failsafe it was bound with), turned transmitter back and throttle rc input jumps very briefly to 1300, motors spin and then throttle rc input returns to 1000.

sirbadger, 2015-03-05:
Any thoughts on this? config conflict or issue?

hydra, 2015-03-14:
Cycle time changes are expectd, it's due to the current beeper implementation.

Failsafe documentation has been updated.  Please read this: https://github.com/cleanflight/cleanflight/blob/master/docs/Failsafe.md

I verfied that the failsafe was working as expected this evening with an SBus RX.

Please re-open if you think there is still a bug, however from my testing and what you've written I'm failing to understand what the issue is here.  This did prompt me to document the failsafe system a bit more so some good has come out of this discussion.

closed 2015-03-14 by hydra