cleanflight/cleanflight #580: MTK GPS

zdar, 2015-03-03:
I'm thinking about adding MTK GPS. I have code mock-up in my fork
https://github.com/zdar/cleanflight/commit/99f65e883589b14be7ef50c7779f06cda5ea3671

It is completly untested now. Before I continue with it into usable form (and pull request) - Is proposed solution/code style acceptable?

thenickdude, 2015-03-19:
That code looks like a good solution to me.

What exactly is an MTK GPS, what are the common models, and do they not also support a NMEA mode?

zdar, 2015-03-19:
There is a bunch of GPS modules based MTK chipsets. I personnaly have Quectel L80 which is waiting to be soldered to & tested with the flip32+ board.

MTK - Basicaly they are NMEA compatible GPS (they output classic NMEA strings) with some proprietary commands for configuration (https://www.sparkfun.com/datasheets/GPS/Modules/PMTK_Protocol.pdf)

hydra, 2015-05-15:
Code available in #729 

Please test it and provide feedback on the PR.

closed 2017-02-25 by hydra