cleanflight/cleanflight #584: CC3D becoming unstable mid flight

l2smith2, 2015-03-04:
I've recently put cleanflight on my CC3D board which was performing well using Openpilot firmware.
I switched because I wanted to use oneshot on my new Kiss escs and I don't like the GCS GUI.

General hovering and flying around all seemed well but when flying much faster but no rapid stick inputs in Angle mode the miniquad became very unstable. Seemingly trying to level itself at changing angles then just flipping out of control to the ground. 

Props were new no vibration issues, CC3D board mounted double sided foam as with stock firmware.
P gain was reduced to 2.5 to stop fast oscillation
PID controller was multiwii rewrite.
I've had no such uncontrollable issues with the stock firmware or cleanflight on my naze32 board.

thenickdude, 2015-03-04:
Is it now doing slow oscillations? "I" set too high can cause slow oscillations.

l2smith2, 2015-03-04:
I lowerd P because of very fast oscillation its pretty much gone now. no slow oscillation. 
It was flying really well then after 1 min pitched back real hard then forward then side to side then just every which way until it hit the ground. I had pretty much no control from it starting to change attitude

hydra, 2015-03-04:
Does it fly OK in acro/rate mode? I.e. no horizon, no angle mode.  I saw a similar behaviour in auto level modes on a cjmcu quad just by changing the props.

l2smith2, 2015-03-10:
I'll have to wait for more propellors to arrive to test it some more.

l2smith2, 2015-03-13:
Should have propellors by Monday for more testing

l2smith2, 2015-03-30:
i'm waiting on a new motor for that frame now. used too long a screw
rebuilding and damaged the windings.

I have found that I can't update the board though. I tried flashing like
the first time with the bootloader pins and an ftdi adapter. also just the
usb plug. can't connect in any combination for flashing/updating

The instructions in the documentation aren't very clear either in a step by
step way.

hydra, 2015-05-15:
@l2smith2 I will close this issue.  If you're having issues reflashing ask in the irc channel or look for videos.  e.g. https://www.youtube.com/watch?v=DRm_CyAyKcs

Please re-open if there is still an issue.

closed 2015-05-15 by hydra