cleanflight/cleanflight #586: no servo output on cc3d with TRIcopter mode

lijingwei0710, 2015-03-04:
flashed the 1.7.2 firmware to CC3D. and using mix tri ,the servo output is S3,but can not found s3 output on CC3D.

hydra, 2015-03-04:
The pwm mappings and timer allocations needs updating for the CC3D.

catchthewind, 2015-03-22:
Is it already fixed?
 I have uploaded cleanflight 1.8 to my CC3D but yaw servo is not working at all :(
 I am also using tricopter mix.Tail servo is S1.

catchthewind, 2015-03-22:
Is there any way to manually map servo to output 4,5 or 6 or get any other way of controlling it?

rcaridade, 2015-04-09:
I've a tricopter too and can find where to connect servo 1! Nor out 4, 5 or 6 work.

I don't know what to do.

catchthewind, 2015-04-10:
You need to look for someone who can fix it or wait for project devs to do it..

rcaridade, 2015-04-10:
I'm a computer programmer and trying to solve this problem, but it's taking a lot of time to understand the code.

catchthewind, 2015-04-10:
Due to this and other problem #706  I have changed my controller to flip32 which is better supported. i still have CC3D waiting for these problems to get fixed :)

CopterFail, 2015-04-11:
CC3D and cleanflight 1.8.1 -> still no output for servo S1

nickaus, 2015-04-12:
I've just brought a CC3D for a Tricopter.
 Spent hours trying to get the tail servo to work then I find this thread.
 Not happy - What are the chances of getting this fixed quickly ?

CopterFail, 2015-04-12:
Did anyone know, what the last cleanflight version was that was working for a tricopter? Meanwhile I'll try openpilot...

hydra, 2015-04-12:
Don't think its ever been supported or tested on the CC3D. The CC3D timer layout doesn't make things easy, code-wise.  Check pwm_mapping.c and timer.c

CopterFail, 2015-04-13:
Ok, I've checked out the 1.8.1 and changed the pwm_mapping of pwm12 to servo. The output is on channel 6 but the pwm is 400Hz and oneshot is a bad idea (the servo seems to be handled as esc).

Redshifft, 2015-04-14:
Well, Well, no Yaw servo on Naze either untill you open the throttle.....You try that ?
1.8 is the same ! can we get this fixed ?

rcaridade, 2015-04-14:
Does any one knows if it will be a solution to S1 in CC3D, I was trying to recycle this board and bought Mag and Baro to have better flight.
But unfortunatly Can use it on my tricopter because there is no servo.
Is this going to be fixed on the next version? Or it's better to forget it because there's no solution?

catchthewind, 2015-04-14:
@rcaridade buy flip32 10dot and you are ready to go :)

rcaridade, 2015-04-14:
@catchthewind  I know that is the solution, I've two APM and two Flip32+, but as I told you I was only trying to recycle it.

catchthewind, 2015-04-15:
@rcaridade You can use it with openpilot. I went back on one of my cc3d and its really OK.

CopterFail, 2015-04-16:
@hydra : Hi, I think this is a "bug" , no "enhancement". Otherwise it should be documented that a CC3D can not be used for tricopters.

catchthewind, 2015-04-16:
True!

rcaridade, 2015-04-16:
I would be very please if some one solved it. I'm hoping that it's a simple thing to solve, this way no one has to waste too much time to fix it.
Thanks for your good work.

hydra, 2015-05-15:
I changed it to bug as requested, but then changed it back to enhancement because it was never working, development happens iteratively; otherwise everything that's not done yet would be a bug and not an enhancement.

CopterFail, 2015-05-15:
Ok, you are right. But many people know baseflight and cleanflight as a good choice for tricopters. cc3d seems to be supported and is very cheap. So many guys (often beginners) start to build a tricopter project and run into trouble. Maybe another of thoses "cc3d/tricopter issues" will be started...

catchthewind, 2015-05-16:
I think it should be a BUG and sorted quickly. Otherwise you should add info to CC3D readme stating clearly that CC3D does not support tricopters while using Cleanflight...

JohnOCFII, 2015-05-16:
@catchthewind The CC3D documentation already states that tricopter support is untested.  Based on the notes above, this does not appear to be a simple fix.

catchthewind, 2015-05-16:
@JohnOCFII   support untested is not the same as NO support. Tricopter needs a servo. No servo - no tricopter. Documentation should say Cleanflight for CC3D does NOT support tricopters at all.

sreichholf, 2015-06-02:
I may want to rethink your understanding of "untested" instead of flaming people here for no reason.

For the average Developer untested is totally equal to "Never tried, no idea if it works at all".
And from my personal experience untested things (almost) never work the way they're expected to.

catchthewind, 2015-06-02:
I am so sorry. I did not intend to flame anyone. I was just sad after spending countless hours on making it work. Now we can say it is tested on a tricopter and not working. Saying no tricopter support would save time for others.

CopterFail, 2015-06-19:
Any news with this issue? Did anybody try 1.9?

fhorinek, 2015-06-22:
I made modification to mapping for CC3D and now it is working!
but it is not yet merged to master, I just made pull request
https://github.com/cleanflight/cleanflight/pull/1042

CopterFail, 2015-06-22:
fhorinek, you are my hero. It works and the YAW servo is displayed in the configurator tab. Min/Max limits  in this tab are 1020/2000 (the default values from rx.h , #define DEFAULT_SERVO_MIN 1020 )

renatoa, 2015-09-18:
So this is still not fixed even today ? 
Any technical reason for this, should I wait a reasonable time or go back to openpilot ?

sppnk, 2015-09-18:
Not yet. It seems CC3d is a real problem with cleanfligt : #1076

CopterFail, 2015-09-21:
It is fixed and works fine, see #1042. But the pull request is still not included... You've to add the 2 line on your own.

renatoa, 2015-09-22:
Add and compile, not having all the required environment, please, is anyone kind to make a hex for me ?
Thank you !

a6j, 2015-11-05:
The same would be grateful if someone will bring * .bin firmware

renatoa, 2015-11-05:
Because nobody did it, finally I built it myself. works... :)
No idea where to post, I can email you.
The build is based on the version committed before September 26th when I downloaded the sources.

nathantsoi, 2015-11-05:
I fly betaflight on my tricopter. The name is misleading, it is not beta. It is amazing, the FIR filter and loop sync'd to gyro makes it fly so, so well. https://github.com/borisbstyle/betaflight/releases plus, there is a bin for the cc3d :)

a6j, 2015-11-05:
nathantsoi, tried firmware beta flight CC3D.bin 2.1.0 and 2.1.1, the result is negative. The board CC3D servo is not working.

renato, if you do not complicate please skinte firmware badmentonych@ya.ru

renatoa, 2015-11-05:
Betaflight, another spin of naze ?!!! :o  is there a compare of features somewhere ?

a6j, sure, will email the build asap

a6j, 2015-11-05:
Many thanks to Renato! Servo works!

renatoa, 2015-11-05:
:)

EAM5892, 2015-11-12:
Renatoa could you explain me how you did it, please?

My email is eam5892@gmail.com

renatoa, 2015-11-12:
Checkout, changed as posted below, and recompiled:
https://github.com/cleanflight/cleanflight/pull/1042

EAM5892, 2015-11-12:
Hi renato, thanks for the reply,

I already saw this thread (#1042), but the problem I have is that I don't
know the procedure of recompiling and updating it in Cleanflight.
Could you explain it step by step, please?

renatoa, 2015-11-12:
Is the procedure explained here:

https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Windows.md

but is simpler to send you the bin, trust me :)

kuson, 2015-11-13:
@renatoa appreciate if you can send me the bin to kusons @ gmail. com
I can also share it on Dropbox for you if you want :)

EAM5892, 2015-11-13:
Could you send it to me too, please?

Thanks in advance

renatoa, 2015-11-13:
What email, please ? 
You can PM me, I think, instead posting here, spam robots are harvesting everywhere :)

EAM5892, 2015-11-13:
eam5892@gmail.com please

a6j, 2015-11-13:
I compile the firmware fix for tricopter version 1.10.0 (9f95334).
It is that all mail not to throw.
https://yadi.sk/d/AB6bMsJOkSELZ

kuson, 2015-11-14:
Thanks Renatoa!  BTW, which Output do I use for Servo 1 on CC3D -- Output 4?  I should flash it from CC3D rescue, and use the bin file right? Thanks ;)

renatoa, 2015-11-14:
Yes, and yes :)

kuson, 2015-11-16:
@renatoa, @a6j   I got the firmware flashed.  However, the whole frame goes spinning (if I remember it turns clockwise).   I was under the front CCW+CW and back CCW (standard to CC3D / PixHawk) so I switched to all CCW, and it was still  the same.  Currently if I yaw left on the tx, the servo turns the motor left, and ifI yaw right, the motor turns right.   What I can think of is what I read somewhere that I have to reverse the Servo Direction.  Is this done on Servo0 on the GUI or via the command line? Thanks again, and hope this will also help others :)

nathantsoi, 2015-11-16:
@kuson

i had to set this in the cli: `smix reverse 5 2 r`

renatoa, 2015-11-16:
Happened me too, but because wasn't able to reverse it by cli as above, reversed mechanically, changed the servo position, not sure if a solution for you, my mount allowed.

Nathan, in the firmware I built the servo output is pin/ch4, is 5 in your example ok ?

a6j, 2015-11-16:
I set up the server as follows:
set servo 5 1100 1900 1500 -100 -1

renatoa, 2015-11-16:
Again, same 5 ... how are output pins numbered ?

a6j, 2015-11-16:
The servo must be connected to pin 4.
The CLI specify: set servo 5 1100 1900 1500 -100 -1
Servo tab:
servo 5 = 4 pin

kuson, 2015-11-17:
@a6j @renatoa  thanks, it seems better (not exclusively wanting to Spin)  but I can't get the copter up.  I put a 10 or 15 degree tilt RIGHT, to the back prop at the servo Mid-Point.  Is this ok, or it is not necessary?  Also, Must the propellers be all CCW, or is it ok to have  L: CCW, R:CW, Back: CCW?   Thanks

renatoa, 2015-11-17:
I have the front props counter-rotating, and the tail no idea atm :)
The motor is less than 10 degree right, when seen from the back.
The heading has no drift with all on default, but yaw P reduced to 4.

[20 more replies]

closed 2017-02-25 by hydra