cleanflight/cleanflight #590: S.BUS not working

airiclenz, 2015-03-05:
I am not able to get SBUS working with a Futaba R3008SB. I tried it in all possible combinations:

- with and without inverter
- using input 4 (the one under GND) and not working on all other pins as well
- ```feature RX_SERIAL```
- ``` set serial_port_2_scenario = 3```
- ```set serialrx_provider = 2```
- ```feature -SOFTSERIAL``` (and also not with ```feature SOFTSERIAL```)

u8mydog, 2015-03-06:
I'm able to get it working between FrSky S.Bus and Naze32. The only difference with my setup is I have feature SOFTSERIAL disabled. AND, the naze32 does not support inverted serial for the hard uart so I modified my receiver to invert the serial protocol. I am using cleanflight/Naze 1.7.2.

airiclenz, 2015-03-06:
I am also using cleanflight / Naze 1.7.2. 
Because I was not sure I tried it with SOFTSERIAL enabled and disables as well as with inverted signal and non-inverted signal... nothing.

airiclenz, 2015-03-06:
one more thing... it is working without any issues with the latest version of baseflight.

u8mydog, 2015-03-09:
I've done a lot of digging around into this topic but I'm by no means an expert. I don't know how baseflight can do it on naze32 because the baudrate for S.Bus is too high for softserial and the STM32F103xx processor does not have built in support for inverted serial. I did notice that cleanflight has comments that sound like inverted capability is available on USART2 using Pin_2 (PB2 BOOT1 signal). The comment in target.h is "PB2 (BOOT1) abused as inverter select GPIO". I don't know what this means because the Naze32 v4 schematic shows no inverter option on PB2. I wished there was a Naze32 v5 schematic available.

Someone has a blog describing how to get the non-inverted S.Bus signal from a Frsky rx and that's how I approached getting S.Bus to work. 

Something that is weird is the CleanFlight hex file that gets downloaded via the firmware flasher doesn't appear to match the source for the tag v1.7.2. The reason I know this is the src for v1.7.2 is suppose to support Smartport telemetry with telemetry_provider=3 but the CLI only supports a max of 2. If you try to enter 3 in the CLI then dump, the value is set to 2. I recompiled the hex using v1.7.2 src and now the CLI supports telemetry_provider=3.

ledvinap, 2015-03-09:
I did not find any means for SBUS inversion on NAZEv5 when checking PCB.

airiclenz, 2015-03-09:
I don't need inversion on the Naze32 as I can invert the signal with an inverter-cable quite easily (here is the inverter I developed for this: https://oshpark.com/shared_projects/lBozCaxv). 

I am not sure how baseflight does it but it work flawlessly with the pin marked with 4 (UART-2 RX following the cleanflight documentation https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20Naze32.md), the option "Serial based receiver" and receiver type set to "SBUS".

I am just wondering if it is a configuration issue with cleanflight (which I think is not the case) or if there is a bug.

/Airic

hofftari, 2015-03-10:
I have the exact same issue as OP. I'm using a Futaba R3008SB and have created a simple inverter for the signal cable coming out of the S-BUS port on the receiver.

Originally while running Baseflight, I had the signal cable connected to pin 4 on the Naze32, and everything worked flawlessly. But when I flashed it so it could connect to Cleanflight, this suddenly didn't work anymore.

I have tried everything: Removing the inverter, trying all input pins and testing all different serial port configuration options, and all the combinations between these above actions.

One thing I noticed too, was that when I used set serialrx_provider = 2 and hit save in the CLI, it went back to the default. This did not happen when I set it to values other than 2. I also got this when I used the GUI for this setting.

u8mydog, 2015-03-10:
The following works for me:
- external inverter
- SBus input is pin_4
- flash v1.7.2 NAZE 2015-2-17 0:27 via the gui firmware flasher
- CLI
  defaults
  dump
     verify build version Cleanflight/NAZE 1.7.2 Feb 23 2015 / 14:21:30 (afe44d1)
  feature RX_SERIAL
  feature -RX_PARALLEL_PWM
  set serial_port_2_scenario = 3
  set serialrx_provider = 2
  save

Now a dump will reflect the above changes including serialrx_provider=2.

Disclaimer: I'm using Frsky x4sr modified to remove inverter within the rx instead of using an external inverter. If you are using Frsky x4sr and want to use S.Port for telemetry, either modify the rx to remove S.Port inverter or add external logic. And, build 1.7.2 downloaded via the gui does not match the src. Building the src using tag 1.7.2 will create a hex that supports S.Port by using 'set telemetry_provider=3'. If a dump shows telemetry_provider=2, then hex is not correct.

airiclenz, 2015-03-11:
The exactly same setup is not working for me with the Futaba R3008SB (while the hardware is proving itself working with baseflight in the exact same configuration):

- inverter cable
- Naze32 inpout pin 4
- cleanflight 1.7.2 (verified with your telemetry_provider=3 check)
- CLI settings `feature RX_SERIAL` `feature -RX_PARALLEL_PWM` `set serial_port_2_scenario = 3` `set serialrx_provider = 2`

hydra, 2015-03-11:
@airiclenz please dump the R3008SB sbus data using a logic analyzer or your scope and I can take a look and compare to data I have from the other 4 receivers I have tested it with.

The code has been tested with other futaba and frsky sbus receivers and works great. 

Maybe there is something different about the  R3008SB.

hydra, 2015-03-11:
Also please try comparing signal quality after the signal has been un-inverted at the point of connection to the FC.  Maybe your inverter is messing the signal up?

hydra, 2015-03-11:
@u8mydog can you confirm your R3008SB works?  Can you provide a complete CLI config dump so that the others can try it?

airiclenz, 2015-03-11:
The R3008SB  is definitely working - I know that because it works when the Naze32 is flashed with baseflight (together with the inverter cable) but I can check that again. You'll receive the scope dump later today when I am home again - together with a screenshot showing the signal before and after the inverter cable.

airiclenz, 2015-03-11:
Here they come...

trace capture (inverted) for use with Rigol DS1000Z class oscilloscope (which I think you have): http://airiclenz.com/download/R3008SB_wave_capture.wfm

trace capture (inverted) as CSV file: 
http://airiclenz.com/download/R3008SB_wave_capture.csv.zip

Now follow some pictures. If you need the measurements, please check my very first post in this issue.  

- the yellow channel one is before...
- ...the magenta channel 3 after the inverter

A zoomed out screenshot - as you can see, every 5th time there is a little extra packet. Could this be the issue?:

[image]

A zoomed in view to show that the inverter is not messing up the signal:

[image]

I hope that helps - let me know if you need additional data.
Airic

hydra, 2015-03-11:
Thanks, I will digest that. Can you show me a zoomed in view of one of the spikes that occurs every fifth time and also measure the time between the previous frame and the spike.

airiclenz, 2015-03-11:
You can download the waveform with the extra packet here: http://airiclenz.com/download/R3008SB_wave_capture_extra.wfm

The delay between the main and the additional packet is 2ms:

[image]

the packet itself (duration ~ 0.35 ms):

[image]

I reckon the extra data is the RSSI byte.

Airic

airiclenz, 2015-03-11:
....forgot to post the dump files (it worked with baseflight with this configuration):

Baseflight:
http://airiclenz.com/download/dump_baseflight_Feb122015.txt

Cleanflight:
http://airiclenz.com/download/dump_cleanflight_1.7.2.txt

hydra, 2015-03-11:
Can you also confirm if there are any settings you have made on your transmitter that affect the sbus output of the receiver, e.g. sbus mode/speed.

airiclenz, 2015-03-11:
I did not change any settings regarding the S.BUS. I am not sure if this is even possible.

hydra, 2015-03-11:
The extra data that is occuring should be just ignored by the code, since it's not a the correct start byte for a frame.

hydra, 2015-03-11:
What software do you use to open the wfm files?

Can you use whatever software you have available to decode the serial data so that you can just give me the frame data as a binary or text file?

airiclenz, 2015-03-12:
You can save the file on a USB stick and open it with the scope itself.
Unfortunately I have no tools to convert the trace to binary text.

hydra, 2015-03-12:
Ok, this will have to wait a week or so then since I don't have my scope here.  If your scope has rs232 decode available you should be able to use your scope to decode it and display the data, perhaps you can then either take screenshots or save it using your scope?

airiclenz, 2015-03-13:
ok, here is one example packet (baud: 100.000 bit/s, 1 start bit, 8 data bits, 1 even parity bit):

22 Bytes
0F, 00, 24, 21, 5C, 08, 08, 09, 9F, 00, 10, 80, 90, 80, 04, 00, 01, 08, 40, 00, 02, 10

Hope that helps!

hydra, 2015-03-30:
This still needs investigating.  Thanks for the example packet.

hydra, 2015-03-30:
What's interesting is the packet is too short, it should be 25 bytes long.

```
struct sbusFrame_s {
    uint8_t syncByte;
    // 176 bits of data (11 bits per channel * 16 channels) = 22 bytes.
    unsigned int chan0 : 11;
    unsigned int chan1 : 11;
    unsigned int chan2 : 11;
    unsigned int chan3 : 11;
    unsigned int chan4 : 11;
    unsigned int chan5 : 11;
    unsigned int chan6 : 11;
    unsigned int chan7 : 11;
    unsigned int chan8 : 11;
    unsigned int chan9 : 11;
    unsigned int chan10 : 11;
    unsigned int chan11 : 11;
    unsigned int chan12 : 11;
    unsigned int chan13 : 11;
    unsigned int chan14 : 11;
    unsigned int chan15 : 11;
    uint8_t flags;
    uint8_t endByte;
} __attribute__ ((__packed__));
```

are you sure you dumped the entire packet?

dave-dd, 2015-04-03:
I have the exact same problem here. SBUS works with Baseflight but not with Cleanflight (Naze 1.8.1).

I am using a self made inverter cable and plugged directly into the trainer port of a Futaba T14SG (because I am still waiting for the receiver I ordered). FrSky SBUS to CPPM Adapter works with Cleanflight, but I don't want to use it...

ph30n1x, 2015-04-11:
I have just tried to switch over to cleanflight on my Naze32 and i have run into the same issue with serial SBUS. I have tried with cleanflight 1.7.2, 1.8.0 and 1.8.1 with no signals. changing back to baseflight SBUS works.

ph30n1x, 2015-04-13:
I have done a bit of research and found that some people had run into issues using SBUS from the R3008SB with other controllers. The fix for those people was to use Mode D on the receiver. I have tested both Mode B & D with no difference on 1.7.2 and 1.8.1. Again the system works with baseflight.

ph30n1x, 2015-04-13:
Alright, this has been confirmed as a bug. A fix has been applied to 1.9.0 at the bottom of page 2 of the following site

http://translate.google.com.au/translate?hl=en&sl=de&u=http://fpv-community.de/showthread.php%3F61213-NAZE32-s-BUS-und-Cleanflight&prev=search

hydra, 2015-04-13:
Can someone post a diff of the fix?  From my understanding it just looks like some of the futaba receivers don't output as many channels.

PilotMaximus, 2015-04-15:
Has this firmware update been added to Cleanflight configurator/flaser?  I have a Futaba 7008sb and I am having the same issue.  Base flight works fine with inverter supplied by Aerial freaks Mojo 280 with Acro Naze.  Flash to CleanFlight and no inputs from rx.  Flash back to BaseFlight and all works.  Really want Cleanflight though.  Not familiar with manual flashing.  Also can't read German and translator is doesn't work enough to be Comfortable.

hydra, 2015-04-19:
It seems the difference is the futaba RX's above send less channels out and the packet checking never completes.

I think the solution might be to look for frames in a similar fashion to how the PPM code does and calculate the amount of channels based on the length of the frame received.

```For those who are interested, 
the SBus.c file amended as follows: 

#define SBUS_MAX_CHANNEL 14 // 18 

and all channels 12-15 uncommented this twice in the code: 

/ * Unsigned int chan12: 11; 
unsigned int chan13: 11; 
unsigned int chan14: 11; 
unsigned int chan15: 11; * /
```

PilotMaximus, 2015-04-20:
Sorry.  Newb here.  Will this or can this be corrected in CleanFlight configurator software Update?  Not real familiar with the CLI but willing to try if needed.  If the above CLI's are copied and pasted right on the command line will it work or does it have to be pasted directly into the correct line of code via the command line?

PilotMaximus, 2015-04-24:
If someone can help me get this code were its needs to go that would be great.  New but willing to learn.

```For those who are interested, 
the SBus.c file amended as follows:

#define SBUS_MAX_CHANNEL 14 // 18

and all channels 12-15 uncommented this twice in the code:

/ * Unsigned int chan12: 11; 
unsigned int chan13: 11; 
unsigned int chan14: 11; 
unsigned int chan15: 11; * /

PilotMaximus, 2015-05-01:
#define SBUS_MAX_CHANNEL 14 // 18
this command enters with no problem

however, and I fear the most important part is the next bit
/ * Unsigned int chan12: 11; 
unsigned int chan13: 11; 
unsigned int chan14: 11; 
unsigned int chan15: 11; * /

is bombing so I must be entering it wrong.  I copied and pasted with no luck.

Not sure what 12-15 uncommented this twice means.  Can anybody elaborate a little better?

Do i have to dump then paste this into the dump then reapply the corrected dump?

What am I missing?

daniel-maiorano, 2015-05-11:
I have a friend having the same issue. So we put a protocol analyzer behind the SBUS port of a R3008SB.
There is a slight difference between the SBUS packet sent by FASST receivers and T-FHSS receivers.
FASST receivers send a 0x00 at the end of each packet, T-FHSS receivers sends 0x04, 0x14, 0x24 and 0x34 packets alternatively (is more likely a SBUS2 packet but without the telemetry data).

IMHO the way to correct this is to recognize the end of the packet when it matches 0xN4, given N to be 0, 1, 2 or 3.

Example of the last byte of a classis SBUS Packet (on a R7008SB, FASSTest Receiver)
[image]

Examples of the last bytes of a SBUS packet on a R3008SB (T-FHSS Receiver)
[image]
[image]
[image]
[image]

daniel-maiorano, 2015-05-11:
Actually, I think is this part of the code that should be changed
[image]

hydra, 2015-05-12:
@daniel-maiorano perfect, thanks for the protocol analyzer examples, that is just what i needed.

hydra, 2015-05-12:
@daniel-maiorano did you keep the logs from?  I would love to have a copy of some generated from the receivers you have.  I have the same software as you so I can load them here.

daniel-maiorano, 2015-05-12:
Unfortunately no, I didn't kept them. 
And it's not my receiver, it's a friend's one. 
I will see him on Thursday, I can make a capture and then send it to you. But basically, as per my observations, the SBUS frame of a R3008SB is exactly the same than the SBUS2 frame of any FASSTest receiver but without the telemetry data. I don't know if you know how a telemetry data is sent over the SBUS2...
If you want to call me, it should be simpler by phone. 
I send you my number to the email address shown on your profile

daniel-maiorano, 2015-05-13:
About the "little packet" we can see every 4 packets, this is the telemetry data of the Receiver Voltage.
[image]

The SBUS2 protocol splits the telemetry data on 4 blocks of 8 slots, I suppose they do that to avoid locking the bus for a long time. between each one of those blocks, there is a "channel information" packet.

The way to identify if the telemetry data block corresponds to slot 1 to 8 or Slot 9 to 16 or slot 17 to 24 or slot 25 to 32 is the end byte of the "channel information" packet: if it ends on 0x04 it means that the subsequent block refers to slot 1 to 8, if it ends on 0x14 it means  that subsequent block refers to slot 9 to 16, and so on

hydra, 2015-05-19:
I will be building a new 1.9.0RC binary later today that will include a possible fix for this issue.

Please can people report back with:
a) what rx you have tried with the new build.
b) if it worked or not.

closed 2015-05-19 by hydra

Stefan989, 2015-05-19:
I would like to test with the R7008SB but I 'm not a programmer . Can someone post the amended Naze32.hex by mail .

airiclenz, 2015-08-11:
Just for you to know - in case you received no feedback yet - it is indeed fixed! Thank you!!

commit 26f2affd (2015-05-19): Ignore SBus end byte checking, this should improve compatibility with
some Futaba SBUS RX's that send telemetry data after SBus frames.  Fixes
#590