cleanflight/cleanflight #595: JR XG8 XBus channel mapping incorrect

andypnz, 2015-03-06:
Using a JR XG8, XBus mode B, 731BX receiver.

Using JR/Spektrum channel mapping......

Tx pitch = roll in cleanflight
Tx yaw = pitch in cleanflight
Tx roll = throttle in cleanflight
Tx throttle = Aux 1 in cleanflight
Tx Aux 1 = yaw in cleanflight

Maybe we need custom channel mapping?

Thanks
Andrew

Pierre-A, 2015-03-07:
**andypnz**, did you try the *map* command in CLI ?

andypnz, 2015-03-07:
Thank you! Was unaware of that command...

AER2T134 and all is perfect :-)

3daerovision, 2015-09-28:
Hi andypnz, how did you connect the RA01T to your board ? Did you use custom firmware ?
Thanks :)
Tristan

closed 2015-09-28 by andypnz

andypnz, 2015-09-28:
Hi Tristan

I happen to be working on a 250 racer manual, and describing DMSS satellite hook up. Here is the section of the manual: https://www.sendspace.com/file/n0gq51 (I could not attach a pdf here).

DMSS stats or XBus supported by Cleanflight (certainly 1.9.0 on) and Betaflight.

You will need custom channel assignments:
For JR DMSS satellites, assigning the channels as AER2T134 should work, but do check. Remember to click save after changing the assignment.
-	Click on the ‘CLI’ tab
-	In the box which says ‘Write your command here’, type ‘map AER2T134’ and hit enter
-	Now type ‘save’ and hit enter
-	The FC will now reboot
After 30 seconds or so, click on the ‘Receiver’ tab again.

ISSUES:
Not all sats work! 
- I have a older (dark brown/black PCB) RA01T which does not work
- Newer (white PCB) RA01T does work
- Latest RA02TL does not work.

Unsure why. I am hoping to get some info from JR, and then post an issue on this.

Andrew