cleanflight/cleanflight #597: LED Strip with PWM receiver

RC2211, 2015-03-08:
Hello guys. First off: thanks so much for the hard work on Cleanflight.

My issue is this: I only use PWM and S.bus Receivers. I recent installed some WS2812 LEDs onto my 250 quad, and found out I need a PPM receiver to support this. 

Could you guys add an option into the firmware where you can choose to have Pin 5 powering the LEDs and all of the other RC pins still supporting PWM?

Thanks for any help!

digitalentity, 2015-03-08:
Hello! I have had this problem as well. Solved it using a cheap pwm to ppm
converter from eBay. Not sure if your suggestion can me implemented in
firmware without much effort.

RC2211, 2015-03-08:
Thats a great idea. I'm a bit tight for space, but it could work. 

I would still like to see someone try to add this to the firmware.

All that needs to be done is isolate RC pin 5 from the others, although I don't know much about coding techniques 

Thanks!

dkisselev, 2015-03-10:
I feel like a good way to handle this might be to set up a configuration page that lets you explicitly see/configure what each pin maps to.

That way you can choose how much PWM you have, how your softserials are laid out, etc. (of course there are technical limitations where some pins are analog in, others are digital, but that would roll in as part of the options that you get for configuring each pin)

RC2211, 2015-03-10:
This sounds perfect, and very easy to use. I'm excited for the release! I wish I could contribute but coding at this level is way beyond my sphere of knowledge

handsomejackuk, 2015-03-12:
had the same issue, and in the end made my own pwm to ppm convertor... real easy cost like £4....and have so many more outputs to use.. i did the same for both my pwm receivers and they have proven to be very reliable too.. i was real surprised they work...

hydra, 2015-03-13:
You would be limited to 4 PWM inputs only on the naze because of timer allocation.  PWM is a bit old-school.  An FC with better pin mappings solves this problem (SPRacingF3). 

Yes, timer/pin allocation needs some work.  I'll close this, but please see #357

closed 2015-03-13 by hydra