cleanflight/cleanflight #598: CC3D with external HMC5883L @Flex Port

Ludn, 2015-03-08:
i did read #525 which should be included in v1.7.2?- not sure if my board GY271 is really ok.
Checked SDA SCL interchange already.

Set up Ports UART3 = unused, set mag_hardware=2.
This is the only way to get after boot up the sequence as follows.
Scaned with a logic analyser, does not look too bad:

Write 0xEE + NAK
Write 0xEE+ NAK, Write 0x3C + ACK, 0x00 +ACK, Read 0x3D + ACK, 0x48 + NACK
followed by silence..
-Should it work in this configuration, may be the pull downs are too strong, any ideas?

thenickdude, 2015-03-09:
It's impossible to set `mag_hardware` = 2 in 1.7.2 using the CLI due to a bug which was fixed in this later commit:

https://github.com/cleanflight/cleanflight/commit/8a541a9fa28197da2d3f2b858ba837be5b9cea1d

Use "dump" to confirm the actual value of your `mag_hardware` setting.

Ludn, 2015-03-09:
Thanks, can't wait getting this brilliant project to work also on other flight controller like sparky or flip32+.

'set mag_hardware = 2' writing on CC3D is confirmed using 'get' or 'dump'.
Indeed other setting does also create I2C communication on flexi port 682ms after boot up as noted above. However 'Mag' is not detected also an additional GPS doesn't help on this.

I have seen some SW-fragments that interrupt connection to DRDY needed, is it?

Unfortunately i am more into HW and not familiar with SW-repository and compiling (yet). I am happy for hints/links to start with to get your commit compiled to a bin-file to test it.

Since i found the HMC5883L driver a was hopping for native compass support. -It' working on the Naze32 version, isn't it? Does a different port make life easier?

Ludn, 2015-03-10:
Did remove 2k2 pull-ups on compass board, still no success :(
Communication still stops after Write 0x3C + ACK, 0x00 +ACK, Read 0x3D + ACK, 0x48 + NACK

Isn't here anyone using a CC3D with a working compass who can give me hint?

ledvinap, 2015-03-10:
Can you post complete communication with HMC5883L ?

Ludn, 2015-03-10:
an overview first:
[image]

in detail:
[image]
[image]
[image]

digitalentity, 2015-03-10:
Communication looks really strange. It seems that magnetometer on your GY271 board is not responding to it's address. Can you post a photo of your GY271 board?

digitalentity, 2015-03-10:
This is how a proper initialization of HMC5883L compass look like on my CC3D + HMC5883L breakout board [image]

ledvinap, 2015-03-10:
0xee is BPM085 or MS5611, NAK is ok if no baro is installed.

Then register 0x0A (ID A) is read, 'H' (0x48) is returned correctly. 

That is last communication?

Where did you get 0x3C + ACK, 0x00 +ACK ?

digitalentity, 2015-03-10:
Oops, sorry, posted the wrong screenshot. This one was for BMP180 baro.

Ludn, 2015-03-11:
@ledvinap
yes, Read 0x3D,  0x48 is the last communication. Shouldn't it end with ACK?
0x3C, 0x00 was the same 3rd and last term. It might changed cause of the missing second unnecessary pull ups.
@digitalentity
could you send the right screenshot or confirm after Read 0x3D + ACK, 0x48 which happens  (NACK / ACK)

ledvinap, 2015-03-11:
@Ludn : Master send NAK on last byte to tell slave that transfer is done, so it should be OK. 

The code is here: https://github.com/cleanflight/cleanflight/blob/3e73b3de5304fd235586831fbcb391ac974ee942/src/main/drivers/compass_hmc5883l.c#L125

Maybe data are received incorrectly? Can you check with oscilloscope?

digitalentity, 2015-03-11:
@Ludn, NACK is ok, communication with my mag looks the same.
[image]

hydra, 2015-03-13:
Perhaps try running some other example code on your GY271 to first verify that the mag is actually working.

Once you've confirmed the mag on your board is working then please re-open this.

closed 2015-03-13 by hydra

Ludn, 2015-03-21:
Thank you to all, wanted to report that issue is fully solved with v1.8.0 cause communication is continued and MAG works well.