cleanflight/cleanflight #600: Autotune yaw not usable in a tricopter

edugit, 2015-03-09:
Reading how autotune works  It only autotunes roll and pitch from my understanding. Yaw PID is a proportional calculations from the other two, and this is calculated for a quadcopter.
I used autotune in a tri. It worked well for pitch and roll but increased a lot the yaw PIDś making the tri uncontrolable in that axis.

Is it posible to disable the yaw PID calculations? or enable autotune only for pitch and roll?

hydra, 2015-03-11:
yes

edugit, 2015-03-11:
Great!!!
I am very happy it can be done.
The way of doing this perhaps is evident but I can not see it.
Since autotune is one of the features that makes cleanflight so attractive is it possible to get a more elaborated response?
Or would it be more appropiate to move this topic to "pull request"?
Thanks anyway for your quick answer

hydra, 2015-10-06:
autotune has been replaced with G-Tune.

closed 2015-10-06 by hydra