cleanflight/cleanflight #603: WHAT do 3 short beeps, repeating, mean?????

kuehn53, 2015-03-10:
I had an issue with CleanFlight and a DragonFly32 Pro (Naze32 clone) constantly issuing 3 short beeps, repeated endlessly, during flight.  Turning off VBAT stopped it; I believe it was caused by having overly powerful motors for the quadcopter (HobbyKing DT-900s).  When I changed motors to Sunnysky Angels, the beeping stopped.  I have searched for a list of beep codes for CLEANFLIGHT but haven't found such a list.  Can anyone tell me what the repeating pattern of 3 beeps was telling me?  Thanks, Brian.

thenickdude, 2015-03-10:
If you had set up a beeper mode on the Modes tab, and you inadvertently
activated it, that makes three beeps continuously.​ I think that's the only
one.

thenickdude, 2015-03-10:
(for reference, the warnings triggered by VBAT monitoring are
short-medium-medium for "warning" level and short-short-medium for
"critical" battery level.)

kuehn53, 2015-03-10:
Nicholas, thanks for the response......I DID (and do) have a beeper mode 
set up on an AUX switch.   Had a flyaway recently, so won't leave home 
without it.   I am certain that the beeper mode was NOT 
activated.....just checked it on the bird with the downsized 
motors.....   the beep pattern is a constant short beep.....beep beep 
beep beep beep beep etc.  There are no triplets...a simple one beep 
continuous pattern when the beeper mode is activated.  The 3 beep 
pattern was SHORT SHORT SHORT ---- Pause --- SHORT SHORT SHORT.   I'm 
pretty sure that the over-sized motors were drawing excess current....  
I changed to a kk2.1.5 board as I thought the DragonFly32 Pro might have 
a problem.    The KK2.1.5 battery monitor beeped (once) every time I 
raised the throttle.  As long as I left the throttle alone, or reduced 
it, there was no sound from the battery monitor.     Those beeps from 
the KK2.1.5 were what got me thinking that the motors were just too big 
for the quad (HobbyKing DT-900's).   I knew they were much more powerful 
than was needed, but they were all I had in the way of 4 matched motors 
when I put the quad together....I thought I could use smaller props and 
go easy on the throttle.   They flew but the throttle was real touchy 
and a constant yo-yo.   I have SunnySky Angel A2212-15 800KV's installed 
now and the DragonFly32 pro is back on board.   That is what I just 
tested for the beeper pattern in beeper mode.  I did fly it this evening 
with VBAT activated.   It was nice and quiet.   I was flying in the dark 
with a headlamp on to illuminate the bird, so I only flew for 7 about 
minutes; a bit over 1/2 of the 3s 2200mah 25C battery pack, so I didn't 
get down to a normal low battery warning signal.  Much friendlier 
throttle with the smaller motors.  Brian Kuehn.

hydra, 2015-03-10:
Beeper system will have an overhaul soon, no point updating the documentation until this is done. 

@kuehn53 Please use the forums or IRC for help/support and not the issue tracker.  the issue tracker is for reporting problems and discussion of new ideas.

closed 2015-03-10 by hydra

thenickdude, 2015-03-10:
I think the beeper mode used to be three short beeps (or perhaps it was three short beeps on Baseflight), because I have it on video here:

https://www.youtube.com/watch?v=R9UGBlraiuk&t=51

However I tested my craft now (Cleanflight 1.8.0) and it did the steady single short beeps that you mentioned. Weird.