cleanflight/cleanflight #604: 100% Servo Throw with only ~40% Rudder Stick

maymayday90, 2015-03-10:
Video:

https://www.youtube.com/watch?v=CyLWtJb_9d4

Servo works fine connected directly to an Rx, same rudder throws. Cleanflight has full throw (1000-2000) in the servo tab.

Worked fine on baseflight about a month ago (will try flash baseflight and see how it works).

thenickdude, 2015-03-10:
The flight controller tilts the servo in order to achieve your commanded
yaw rate. You don't have direct control of the servo (there is no linear
relationship between your yaw input and the servo value).

Because your tricopter is static on the ground, Cleanflight will see that
its yaw correction isn't having any effect, so it'll continue to tilt the
servo further over to try to yaw harder to meet your target, even though
you're not giving it additional yaw input.

maymayday90, 2015-03-10:
Ah, gotcha. And if I lower my Yaw Rate to 0, it becomes much closer to being linear.

I guess I'm used to setting up the board with Yaw Rate at 0 (i.e. new board, new build), but this time it was preset to 0.35 since I was converting from a quad.

Thanks!

closed 2015-03-10 by maymayday90