cleanflight/cleanflight #606: cmix custom for tricopter ?

dronadair, 2015-03-10:
I tried to change the cmix values for my tricopter in the CLI to compensate for non-standard COG position. 
When doing:
mixer tri
cmix 1 1 0 1.586 0
cmix 2 1 -1.352 -0.414 0
cmix 3 1 1.352 -0.414 0
save 

The tricopter seems still flies like with the default values, althought the "dump" command show that my values have been correctly saved.I also tried to put extreme values to make sure. Still the same behavior as default "mixer tri". 

Finally, doing it with "mixer custom" instead of "mixer tri" , and adding the extra line for disabling motor 4: 

mixer custom
cmix 1 1 0 1.586 0
cmix 2 1 -1.352 -0.414 0
cmix 3 1 1.352 -0.414 0
cmix 4 0 0 0 0
save

this does not give me the ability to use the tail servo. So, cannot do a custom mix for tricopter. 

Does cleanflight support custom mix for tricopter? Or does it ignore any changes to the cmix?

Thanks

hydra, 2015-03-11:
this is not something i've looked at, i would imagine that servos are enabled based on the value of the mixer type, let's see if my thoughts are correct...

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L299

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L210

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L232

yup, that's what i thought.

So the answer is no, you cannot currently define a custom mix for a tricopter.

Also, this chunk of code is only executed if the mixer is set to tricopter:

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L424-L435

I think it wouldnt be to difficult to add a `MIXER_CUSTOM_TRICOPTER` setting, do it here:

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L232
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L582
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.h#L48
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L424
https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_cli.c#L147
https://github.com/cleanflight/cleanflight/blob/master/src/main/blackbox/blackbox.c#L321
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c#L320

A pull-request with tested changes would be great!

dronadair, 2015-03-12:
Dominic, thank you for your detailed reply. I will have a look.

Cheers

Raphael

:

hydra, 2015-03-25:
@dronadair Any update?

dronadair, 2015-03-26:
Not yet, i'm traveling, will have a look in a few days.

:

hydra, 2015-06-28:
Provisionally added in 7a1d071efbeda6b27a5bc58dd216d4f66139e513.

See branch https://github.com/cleanflight/cleanflight/tree/custom-servo-mixers

This needs field testing and feedback, build your own binary from the branch.

Example usage:

```
mixer CUSTOMTRI
cmix load TRI
smix load TRI
```

OR

```
mixer CUSTOMTRI
cmix reset
cmix 1 1.000 0.000 1.333 0.000
cmix 2 1.000 -1.000 -0.667 0.000
cmix 3 1.000 1.000 -0.667 0.000
smix reset
smix 1 6 3 100 0 0 100 0
```

hydra, 2015-06-28:
There's also a corresponding configurator branch with the same name if you want to use the servos tab on it.

https://github.com/cleanflight/cleanflight-configurator/tree/custom-servo-mixers

dronadair, 2015-06-28:
Dominic, this is great. Thank you! I got carried away with work so my
apologies for not getting back to you so far.

Will have a look hopefully next week!

Raphael

nsccap, 2015-06-29:
I have a tri with non-std cg which I today compensate with asymmetric prop setup. I'll try to fly this branch on it during the week.

nsccap, 2015-07-01:
I successfully built custom-servo-mixers branch at commit 45cd267 for NAZE (flip32+).
That was the end of success kind of... With gui from the matching branch or normal 0.65.0 everything hung after doing "mixer customtri" and save in cli. That is I could not click any tabs after connect and had to re-flash.

After backing down to 1.9.0 I ran into https://github.com/cleanflight/cleanflight/issues/991 so I didn't get a compatible gui and cf even there.. Backed off to where I started, 4857050

hydra, 2015-07-08:
Yes, GUI is not updated to support that branch yet.  I'm working on the branch more now.

Be aware that index for servo rules and motor mixes via the CLI will be 0 based soon.

e.g.

```
mixer CUSTOMTRI
cmix reset
cmix 0 1.000 0.000 1.333 0.000
cmix 1 1.000 -1.000 -0.667 0.000
cmix 2 1.000 1.000 -0.667 0.000
smix reset
smix 0 6 3 100 0 0 100 0
```

hydra, 2015-07-08:
See #1066

hydra, 2015-07-13:
Done, test flown.

closed 2015-07-13 by hydra

dronadair, 2015-07-19:
Hi Dominic,

Since these updates related to the custom tri, I cannot use cleanflight
from my two macs. One under snow leopard 10.6.8, and Yosemite 10.10
Under Yosemite, the cleanflight crashes after cleanflight, before I can
connect.
Under snow leopard, the "connect" opens the port and gives the id, but
hangs and disconnects. I tried with both USB and bluetooth. I re-installed
the driver, but still no joy. I tried two different controllers, same
issue. It works under windows.
Did you test with your mac setup? Any idea? I cannot test your nice
updates, while i'm finally on vacation with time and space to fly :-) as i
only have my macbook, no windows here :-(

Thanks again for your help.

Raphael

johnty, 2017-04-07:
apologies for resurrecting this thread, but have these changes been incorporated into the current version, or should i still be doing a custom compilation for it to work?

i recently made a brushless tri using a hex-capable brushed naze32 board which made ch1/ch2 PWM outputs assignable to servos since CH3-6 still had mosfet drivers for the motors, unlike the quad-only boards), and the layout wiring was such that m1 and m2 (m0==tail) would have been easier if i flipped their roll contribution. however, trying a bunch of things had no effect at all.

i tried loading the tri and custom tri profiles and then manually changing the mmix values, but all to no avail (flipping the roll contributions had no effect). i ended up just adding some more wiring to hook up motors in their default positions.

anyway, the thing [flies](https://www.youtube.com/watch?v=xnskxcp_0aU) and needs tweaking elsewhere, but at least it works! :)