cleanflight/cleanflight #607: On CC3D LEDs stay dark on selected Profil 2 only..

Ludn, 2015-03-10:
..WS2812-LEDs works fine in profile 1 and 3, but not for 2nd profile.
All Profiles are using PID controller 3.
Quite strange and can't explain why. Here segments of my quite simple configuration:
Profil
# version
Cleanflight/CC3D 1.7.2 Feb 12 2015 / 13:45:03 (d12d195)
# dump master

# mixer
mixer QUADX
...
feature RX_PPM
feature VBAT
feature FAILSAFE
feature LED_STRIP

# map
map AETR1234

# led
led 0 0,0:S:WA:7
led 1 0,1:S:F:0
led 2 0,2:S:WF:0
led 3 0,3:S:TC:0
led 4 0,4:S:WA:0
led 5 0,5:S:C:2
led 6 0,6:E:C:2
led 7 1,4:EU:WC:3
led 8 2,4:EU:AC:3
led 9 3,4:EU:WC:3
led 10 5,4:E:WC:2
led 11 0,0:::0
led 12 0,0:::0
....
# dump profile

# profile
profile 2

# aux
....
set mag_declination = 0
set pid_controller = 5
set p_pitch = 48
set i_pitch = 30
set d_pitch = 42
set p_roll = 45
set i_roll = 30
set d_roll = 42
set p_yaw = 85
set i_yaw = 45
set d_yaw = 23
set p_pitchf = 2.500
set i_pitchf = 0.600
set d_pitchf = 0.060
set p_rollf = 2.500
set i_rollf = 0.600
set d_rollf = 0.060
set p_yawf = 8.000
set i_yawf = 0.500
set d_yawf = 0.050
set level_horizon = 3.000
set level_angle = 5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

thenickdude, 2015-03-10:
I think there is a flight mode switch which can disable the leds, and flight modes are specific to the profile that you're using. Check your flight modes on profile 2 and make sure that a led flight mode hasn't slipped in there.

Ludn, 2015-03-11:
Thanks: indeed i had 'LEDLOW' mode defined in 'modes', but for a small range on channel AUX2.
AUX2 did not vary when changing the flight mode and is pretty stable.

Thus there shouldn't any dependency (between AUX2 and profile no.) it helped to delete  'LEDLOW' mode.
After 2nd try reproducing this issue all 'modes' settings are gone, may be EEPROM saving got corrupted but there is nothing suspicious logged '-- EEPROM saved'. Is it being verified after write? -Not sure what's going on, don't want to open another issue yet.

thenickdude, 2015-03-11:
How were you reproducing it? If you were using the "dump" command and then pasting it back in, it does tend to erase your mode settings if you paste it in all at once. This is because the "profile" command halfway through the dump takes so long to execute that many of the following statements end up getting dropped, including the "aux" ones. If you copy and paste everything up to and including the "profile" command with one paste, then execute that, then paste and execute the rest, everything should restore properly.

hydra, 2015-03-13:
Likely this is a mis-configuration or as nick detailed above or configuration corruption issue since there is no dependency or LED state that needs resetting between profile changes.

Potential for configuration corruption is yet another reason why the CLI is bad.

I will close this now but if you can provide steps to repeat from a fresh flash with full chip erase enabled then please re-open.

closed 2015-03-13 by hydra