cleanflight/cleanflight #609: Failsave does not work anymore in last deveopment build (may be limited to Spektrum Sat only)

MJ666, 2015-03-11:
All details I have published in the following RCG thread:

http://www.rcgroups.com/forums/showpost.php?p=31023971&postcount=4474

This is a serious problem and likely will cause a flyaway. Hope not many users will fly with the developers build.

hydra, 2015-03-11:
Yeah, looks like some investigation is required.  I'll look into it.  Thanks for the test report, much appreciated.

hydra, 2015-03-11:
Likely the spectrum RX driver needs updating a little for when the RX reports a failsafe condition.

MJ666, 2015-03-11:
Here I wrote something we have seen from experience with the Lemon RX DSM2 and DSMX satellite receivers:

http://www.rcgroups.com/forums/showpost.php?p=31027412&postcount=4496

hydra, 2015-03-14:
Did a bit of work on this in 7d6e4aa3900370c8a11a020337ea2079846f3de6.

Please re-test with Spektrum RX and report back.

hydra, 2015-03-14:
More changes done in e08c83fdc9142fd2efdf0098710fc48bf194a464

Please re-test a build that has both of those commits.

MJ666, 2015-03-15:
Will do as soon I will be back from traveling. Thanks.

stronnag, 2015-03-15:
hydra, latest build (2015-03-15) did not fix the issue for spektrum sat.

hydra, 2015-03-15:
ok, this will have to wait till i get back from austria so i can use my test equipment to debug it since the problem is not obvious to me.

hydra, 2015-03-15:
did a code walk in irc, found the problem, fixed by 1e7fb08b9d8afc98945b0eb77b6b55ce55c43888

closed 2015-03-15 by hydra

stronnag, 2015-03-15:
Tested on Spektrum. Failsafe works again.

hydra, 2015-03-15:
@MJ666 @stronnag thanks for reporting and testing.

Just pasted this in the rcgroups thread:

The problem came about due to users wanting this:

https://github.com/cleanflight/clean...mment-75368567

Due to this bug several things happened:

1) all serial rx handlers use the same pattern
2) minor performance optimizations were made
3) sumd rx failsafe was improved (matches sbus failsafe behaviour)
4) spektrum rx code was cleaned up, magic numbers removed.
5) memory usage was slightly decreased.

All in all a pretty good result! This is yet another example of how open source communities help make everything better.