cleanflight/cleanflight #612: Not able to use RX_PARALLEL_PWM mode on CC3D boards

Dvogonen, 2015-03-12:
I am trying to use cleanflight on CC3D boards. I am using receivers without a PPM signal, so I need to input signals via the RC_input connector in RX_PARALLEL_PWM mode. 

According to the documentation, vbat needs to be disabled to use pin 7 as input for channel 4,
Pin 8 will then be used as input for channel 5. If vbat is not disabled, pin 7 is used as battery voltage input and pin 8 is used for channel 4.

The problem is that I can not get pin 7 act as input for channel 4. Disabling vbat has no effect. I have tried this on two different CC3D boards to verify that my board is not deffective - same result. 
I have also tested  connecting different receiver signals to pin 7. On the other 4 channel pins everything works as aspected, but nothing on pin 7.
I have obviously checked the continuity for the lead from the receiver to after the contact. The pin7 cable is just fine.

Do I need to do something else to enable pin 7 that is not covered by the information on https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md, or have I run into a firmware bug in clearflight?

hydra, 2015-03-12:
I'll try and verify this on my one here.

Dvogonen, 2015-03-12:
Just made a new test. I verified that:
* with vbat enabled the data going to pin8 ends up on channel 4.
* with vbat disabled the data going to pin8 ends up on channel 5.
So that part is working as it should.

kristoferahlqvist, 2015-03-12:
I got the exact same issue, no matter what I cant get pin7 working on cleanflight.
Works without issues with openpilot

hydra, 2015-03-13:
Confirmed, i know what the issue is, i will work on a fix soon.

hydra, 2015-03-13:
The root cause of the problem is that the ADC initialsation now happens AFTER pwm initialsisation, due to recent ESC compatibility improvements, the problem was caused by the ADC code reconfiguring the input pin.

With 3b77626dd8aca22283ac637a149fb0783958e846 ADC initilisation for VBat now only happens if VBat feature is enabled.

This again relates to #357.

closed 2015-03-13 by hydra

mcknightly, 2015-04-02:
Thanks for all the hard work on this. I thought I saw this change was going to drop on 1.8.1. Is this not the case? I just tested 1.8.1 and I can get the 5th channel with VBat disabled, but I cannot get my 6th channel to work. I tested on OpenFlight and all channels work.

digitalentity, 2015-04-03:
CC3D only supports 5 ch when used with PWM receiver. If you need more than 5 channels - please use PPM

SSimonTemplar, 2015-04-04:
Hi All, 
Disclaimer: CleanFlight total newb. 
Is this #612 issue what is causing me troubles: I have tried 1.8.1 and 1.8.0 both with the same outcome.

http://www.reddit.com/r/Multicopter/comments/31du9q/question_cleanflight_channel_mapping_going_wrong/

I just install cleanflight 1.8.0 in a cc3d. I have a taranis radio with a v8fr-ii receiver. All the channels were mapped correctly in Openpilot. Now, in cleanflight: Aileron and Elevator are fine but Yaw in the radio was moving the throttle bar. I changed the map from AETR1234 to AERT1234 and now Yaw is fine but the throttle bar is controlled by the aux slider in my radio. The throttle stick in the radio does not move any bars, and I had another aux channel with a 3 position switch that also does not move any bars.
I do not understand this or how to change it. I also could not find any documentation on the "map" command. I tried an AERT1235 but the CLI does not let me have numbers higher than 4.
Also, every time i save, the board reboots and i cannot reconnect unless i disconnect and reconnect the usb cable.
This is my dump: http://pastie.org/10072930

xRadx, 2015-04-08:
I am going to come back with PWM RX channels. I just loaded FW 1.8.1 and can see only 4 channels available (and 5 with VBAT disabled). In my opinion there is no need for VBAT functionality for CC3D. I believe this is in most cases taken by OSD or battery onboard alarm. Therefore having 5 or even 6 PWM inputs would be better advantage.
Just for curiosity: atm, what is used the 6th input for?

Donykurniawan, 2015-10-15:
its seem like my problems , with RJ01 xbus, thottle stick move in AUX1 bar. even i change in AERT or AETR, everything else work fine. only throttle problems, when stick move AUX1 bar change, and When AUX1 swt move thottle bar change in clenflight.  my radio JR XG11. i cant change divice AUX1 to thottle. any body knows how to fix this problems.. thank you

uusitalo, 2015-10-19:
having same problem :(

krzysztofkuczek, 2015-11-10:
Hi,
i have same problem with 1.10.0 and rx_parallel_pwm. Chanel #5 not working :(

gandyman, 2015-11-21:
I had a course the same because clean flight is focust on ppm signal. I solve this before on my pixhawk with a pwm to ppm encoder and have done the same now with clean flight on my cc3d.
HK (and others del these encoders for a few buck
http://www.hobbyking.com/hobbyking/store/__62753__PPM_Encoder_Module_HKPilot_32.html
Hope this will help in the mean time

Donykurniawan, 2015-11-22:
Yes,gandyman. When I use rmilec v2 ppm converter, everything fine. But this not good for flying quadracing 250. To much hardware and so heavy..😌

gandyman, 2015-11-22:
> ---------oOOo--(_)--oOOo-----------
> Voor elk probleem is een oplossing.
>    Dus dat is het probleem niet.
> De oplossing. Dat is het probleem.
> ---------oooO-------Oooo-----------
> Deze mail komt van een veilige Ubuntu Linux-computer.
> This mail comes from a save Ubuntu Linux-computer.

I do understan your argument, but there is always a but.... :)

If you remove the connectors on the encoder and solder direct the wires 
on it , as short as possible and glue the encode on your reciever then 
you have only 1 wire from your reciver to the pilot with one connector.

Wen we talk in gram I am sure you are lighter with the encoder then now 
with all the wires and connectors.
Give it a try and weight it all ..

:P

Sccs

Gandyman

Bibabun, 2016-05-08:
Have the same problem with throttle not working in PWM on my RX FS-IA6B, cc3d.
Fixed it by mixing channels or swapping connectors. But still not satisfied i have one AUX channel lost in the end.

commit 3b77626d (2015-03-13): Only enable VBat ADC when VBat feature is enabled.

Fixes #612