cleanflight/cleanflight #614: Add support for APA102 / APA104 LED strips

emdzej, 2015-03-12:
I have just found out about a new type of addressable LEDs. Compared to WS2812 neopixels these seem to be less difficult/resource consuming to handle as they have dedicated clock signal - so no more. Probably they could be easily programmed using SPI.

hydra, 2015-03-13:
Can you provide links to:
a) data sheets
b) products that people can buy

IMHO Since the WS2812's are so readily available in so many different form factors and only require 1 hardware pin there is no current need to support any other kinds of LEDs at this time though.  That said, provide more information and maybe someone will pick up the challenge.

emdzej, 2015-06-03:
Happy to do so:

Datasheet can be found at: http://www.adafruit.com/datasheets/APA102.pdf
Diodes are widely available in various form factors as WSs are: http://www.aliexpress.com/store/group/APA102-APA104-addressable-pixel-strip/701799_256700190.html

You're absolutely right that WSx require only one pin which is quite efficient, but because of strict timing requirements it may be more efortless to support APA ones.

ValkyrieFPV, 2016-02-27:
Any progress on this? I bought a bunch of these LEDS on ebay thinking they were WS2812B but turns out they were mislabeled APA104. would love to be able to use them. all I get when connected to CF is solid blue.

closed 2017-02-25 by hydra