cleanflight/cleanflight #617: [enhancement] Add a watchdog timer

digitalentity, 2015-03-13:
If a flight controller hangs mid-flight many bad things can happen. A watchdog timer would be very useful for rebooting a FC in such case. STM32's backup SRAM can be used to hold current system state (flight mode etc), so that FC can quickly regain control of the aircraft.
This, however, should be distinguished from a normal reboot or a power cycling.

dascons, 2015-03-13:
I feel like it will be a mission to code but is necessary as features and bugs come on board

digitalentity, 2015-03-13:
I am not very well into the internal architecture of cleanflight, but it occures to me we should only preserve most (not all) flagn in the runtime configuration:
- arming flag
- flight mode flags (probably no need to preserve these as they will be almost instantly set by receiver)
- state flags (are they needed to be preserved?)

Probably we can only preserve the 'armed' or 'disarmed' state. If we are disarmed - just do a normal reboot, we are sitting on the ground, no need to restore anything. 
If we are armed - watchdog was triggered mid-flight and we should re-initialize whatever we can as fast as we can to prevent crash.

ledvinap, 2015-03-13:
Currently it takes 30s to calibrate gyro and initialization code is not designed to run quickly... This will require quite large redesign on code.

dascons, 2015-03-13:
Maybe it would be possible to keep last gyro config in SRAM so it can just restore, even loosing features after a fail would be better than dropping out of the sky

digitalentity, 2015-03-13:
Gyro calibration data is fairly small (only 6 bytes), so it can be stored in backup SRAM. STM32F103 processor has only 20 bytes of backup memory available, so we should carefully choose what we should backup. Accel and Mag are calibration data is stored in EEPROM, Baro is calibrated on the fly, GPS is rather independent from FC. I figure we should only backup gyroZero and armingFlags and omit time-consuming initialization when recovering from failure. This would probably meen disabling some features, but as @dascons mentioned it is better than a crash landing.

digitalentity, 2015-03-13:
I'll look into implementing this feature in my fork.

digitalentity, 2015-03-14:
Did some profiling of init() on my CC3D + mag + baro on I2C.
sensorsAutodetect() takes 0.69 sec,
compassInit() takes 1,26 sec.
whole init() takes as much as 3.22 sec.

By omiting delays and disabling features & sensors I was able to get a flyable system state in 0.95 sec. Further speed-up can be done by omiting sensorsAutodetect() since when whatchdog is triggered we already know what sensors we have and how they are configured.

Djhg2000, 2015-03-14:
I really like this idea, mainly because I find autonomous recovery systems fascinating. If we strip out the magnetometer and gyro in sensorsAutodetect() and combine it with the accelerometer portion of reconfigureAlignment() we could trade accurate rotational data for initialization speed.

The accelerometer will probably do good enough for locating the earths center of gravity and then we could use this to do some rough auto leveling until we hit the ground. Come to think of it, since we can assume that we're falling out of the sky, we should probably stop the motors when we detect a fairly large spike in acceleration (magnitude > 50 m/s^2 or something) to protect any live victims.

This feels like some pretty major work, but I'd be happy to help out where I can.

digitalentity, 2015-03-17:
I'm currently working on speeding up gyro init. The only board I have is a CC3D (MPU6000 on SPI) and TauLabs somehow manages to initialize both acc & gyro on that chip in less than 100ms. I think CF gyro and accel initialization code needs to be optimized for speed before we'll even try to create a working in-flight failure recovery. I think we shouldn't trade gyro for speed - gyro's very fast reaction on small rotation angles is critical for stabilizing an aircraft.
I am open to any suggestions on speeding things without sacrificing basic functionality.

thenickdude, 2015-03-17:
The gyro can't be calibrated in flight, as the craft will be rotating and a
zero point can't be determined, so the speed of gyro calibration doesn't
seem to matter?

digitalentity, 2015-03-17:
Calibration data can be stored in backup memory. I am concerned about how
fast actual hardware is initialized to the state it can provide valid (yet
uncalibrated) data.

>

-- 
С уважением, Константин Шарлаимов

Regards, Konstantin Sharlaimov

ledvinap, 2015-03-17:
From simple grep for delay:
battery.c: 23ms
display.c: 400ms
accgyro_spi_mpu6000: detect (150ms - 750ms)+200ms, init 300ms
flash_mp25p16: 50ms
adc_stm32f30x: 10ms
barometer_bmp085: 20ms
compass_hmc5883l:    150+500+500+100ms (2x500ms is callibration)
accgyro_l3g4200d: 130ms
compass_ak8975: 50ms
accgyro_spi_mpu6500: 200ms
accgyro_l3gd20: 100ms
barometer_ms5611: 10ms
accgyro_mpu6050: 135ms
accgyro_mpu3050: 50ms
accgyro_lsm303dlhc: 210ms
naze HW detection: 50ms
vcp: 200ms
main.c: 100ms + blinking

F984dude, 2016-08-16:
This issue is quite old, but as far as I see in the current code, no watchdog has been added.

In my opinion, even a watchdog which just shuts down all the motors is better than no watchdog.
While playing around with higher baudrates for the MSP, I had the FC hang multiple times. It won't respond to anything, and if the motors were spinning, they just continued at the PWM at which the FC started hanging.

This was clearly my fault since it didn't hang a single time at 115200 baud. 
But still, if the FC hangs inflight at a high throttle for some reason, one can only hope and pray that the craft loses stability and starts plummeting to the ground before flying out of sight. I consider just turning the motors off in such an event so be safer for the craft, the people around and the pilot's nerves ;)

digitalentity, 2016-08-16:
@F984dude hmm... good point. I didn't look at it from safety point of view. Worth implementing.

closed 2017-02-25 by hydra