cleanflight/cleanflight #619: Usage of backup registers on STM32F103 targets?

digitalentity, 2015-03-13:
Do we really need to use backup registers in isMPUSoftReset function (system_stm32f10x.c)? 
Following code is faster, cleaner and more reliable:
```
    if (RCC->CSR & RCC_CSR_SFTRSTF)
        return true;
    else
        return false;
```

digitalentity, 2015-03-13:
I believe the code, that uses backup registers was ported from baseflight [https://github.com/multiwii/baseflight/blob/master/src/drv_system.c#L209]. 
According to ST's reference manual, STM32F1xx family has the CSR and SoftReset flag. Using those to detect a soft-reset is a much cleaner way than magic values in backup memory. I've created a PR #620 with the change.

hydra, 2015-03-14:
What about the F3 targets?

digitalentity, 2015-03-14:
F3 targets are using this approach in isMPUSoftReset() already.

closed 2017-02-25 by hydra