cleanflight/cleanflight #631: Failsafe not working (and settings not present in backup)

skyflyer, 2015-03-17:
## Failsafe not working

Using Cleanflight 1.7.2 with Naze32 Acro (without mag + baro). I set the failsafe on the FC using the following values: `failsafe_delay = 10`, `failsafe_off_delay = 10`, `failsafe_throttle = 1100`. 

### Test scenario

1. Turn on TX
1. Connect battery, rx (FrSky D4R-II, using CPPM, telemetry via softserial 1 (not sure if it matters))
1. ARM
1. Add throttle (I used apx 3/5)
1. Disconnect RX from naze

The motors will keep spinning at the same speed as before. Even after 20 seconds (which should be the default `failsafe_off_delay` value.

## Failsafe settings not present in backup

When I backup the settings using Cleanflight Chrome app, only one failsafe setting is written to backup json file -- `failsafe_throttle`.  Other values are missing. They are visible in the CLI `dump`, however.

I'm sorry if this is already known issue, but I wasn't able to find this reported in the issue list.

NB: failsafe on the RX works, but that has nothing to do with FC failsafe.
Complete configuration is [here](https://gist.github.com/skyflyer/7568ebd050605cebca6b).

skyflyer, 2015-03-17:
OK, (banging on my head). The FAILSAFE feature must be enabled in order for the failsafe to work. Sort of related to #488, I think this feature should:

1. Be on by default
1. Be documented in Failsafe.md - somewhere near the top of the document

I will prepare a pull request with this change later.

skyflyer, 2015-03-19:
Here is a commit that changed documentation of failsafe feature. Do you want me to create a separate pull request for this?

I didn't get around to fixing the backup feature of Cleanflight GUI.

hydra, 2015-03-21:
There us another issue already open for changing the defaults. Please search before opening new issues. See #488

skyflyer, 2015-03-21:
Dominic, #488 is related to changing the default values. This one actually addresses two separate (but related) issues:

1. that `feature FAILSAFE` is not mentioned in the documentation (pull request fixes that)
1. that not all settings are present in the "Cleanflight GUI backup" (json file)

Would you consider reopening it?

hydra, 2015-03-21:
I'll merge the or. Please open the configurator issue in the configuration issue tracker.

closed 2015-03-21 by hydra

hydra, 2015-03-21:
I can't find a PR for the commit that updates the documentation

skyflyer, 2015-03-21:
I just referenced the commit in the fork (I asked whether you want a separate pull request). I'll create it right away.