cleanflight/cleanflight #632: 1.8 flip32

tbrady420, 2015-03-17:
with new update ports 3-4 do not work have confirmed on two boards and different receivers.

thenickdude, 2015-03-17:
By port do you mean PWM receiver input pins? Double check that your channel mapping is correct on the receiver tab, and double check that you've got the "PWM" receiver type selected.

Receiver pins 3 & 4 are used for UART2 when configured to use a PPM receiver, so this could be a bug where those pins are wrongly assigned to UART2 even in PWM receiver mode.

tbrady420, 2015-03-17:
yes pwm. Cannot save changes to board or load backed up setting completely. every time I hit save in any tab it doesnt write to the board. under ports it has 3 urat1 listed all with msp, telem etc. when changing to unused and saved nothing happens. all tabs loose save/reboot function. I have tried 2 flysky recievers and have tried the board completely unplugged. Love clean flight just wanted to report an issue

thenickdude, 2015-03-17:
That sounds like you aren't running the latest configurator, Configurator 0.63.0 is required for Cleanflight 1.8.0. Follow these instructions to upgrade:

http://www.rcgroups.com/forums/showthread.php?t=2249574&page=319#post31078207

tbrady420, 2015-03-17:
thanks you that fixed it. I cant believe I over looked that in the firmware release at the bottom. I was just to excited about the update. Thanks again

thenickdude, 2015-03-17:
Did it also fix the issue with channels 3 & 4?

tbrady420, 2015-03-17:
no i still have issues its set to pwm and channel 3 is mapped to aux 2.  I have no yaw or throttle. only ports active is urat 1 for msp.

thenickdude, 2015-03-17:
I'm assuming that your channel mapping is currently the default TAER1234? If not please say what it currently is on your receiver tab.

If channel 3 is ending up on aux2, you can change the mapping on the receiver tab to TA2R1E34 to swap elevator and aux2 around. Or just switch those wires around from your receiver.

icdark, 2015-03-18:
I also have the same issue. I have Taranis with settings exact as in 1.7.2 and it works there. I have 0.63.0 Configurator and I couldn't save receiver settings.

tbrady420, 2015-03-18:
I got it working by following the steps to get dev GUI. Make sure you are launching 0.63 not 0.62 because you have two versions of clean flight GUI now installed on chrome.

closed 2015-03-18 by hydra