cleanflight/cleanflight #637: Improvements to the CLI - ignore white spaces

cosmic2112, 2015-03-17:
When I am restoring the configuration after an update using the captured data from a dump command I always get errors as shown below below and others too but some don't (like the map command).

Just to clarify - if I type the commands they work fine so the only difference seems to be the fact that there are SPACES after the command which get captured during the dump command.

So if I type feature servo_tilt+SPACE it will fail.
------------------------------------------------------------------------------------------------
Entering CLI Mode, type 'exit' to return, or 'help'

# mixer QUADX                                                                   
Invalid mixer type    

# feature RX_PPM                                  
Invalid feature name

# feature MOTOR_STOP                              
Invalid feature name

# feature SERVO_TILT                              
Invalid feature name

# version 
Cleanflight/NAZE 1.8.0 Mar 16 2015 / 23:07:13 (8e65707)
#

thenickdude, 2015-03-18:
There aren't any spaces printed after those lines with the dump command. What tool are you using to capture the result of the dump? Are you pasting it into something else in between?

It does seem like ignoring trailing whitespace on lines would be useful, I'll submit a pull request.

cosmic2112, 2015-03-18:
I am using a serial connection using Linux and minicom, I highlight the output of the dump command in minicom and copy and paste that into a text file.
When I highlight the lines to restore from the dump text file and paste it back into the CLI, the capture process adds some trailing white spaces and so some commands fail when you try and restore you're dump.

I hope that explains the problem better.

PR #640 (2015-03-28): Ignore trailing spaces on CLI commands
Closes #637
closed 2015-03-28 by hydra