cleanflight/cleanflight #638: Baseflight 1.8.0. Can not save values for Naze32 board

hawaiianer, 2015-03-17:
Hello,
I just got a new Naze32 (Clone) from Ebay and flashed the 1.8.0. Cleanflight on it. 
I can not save any values (configuration or reciever or modes) with this version. With 1.7.2 it is working.
(I have two of those boards, both have the issue)

Is this issue know or am I the only lucky one?

Regards,

Luz

rocky77, 2015-03-17:
Hello,  yo need cleanflight-configurator v0.63

--- Message initial ---

https://github.com/cleanflight/cleanflight/issues/638

hydra, 2015-03-17:
Closing, please read release notes.

closed 2015-03-17 by hydra

thenickdude, 2015-03-17:
That sounds like you are not using the required development version of the
configurator (0.63.0). Please follow these instructions to install it:

http://www.rcgroups.com/forums/showthread.php?t=2249574&page=319#post31078207