cleanflight/cleanflight #639: Tricopter Yaw Servo direction change possible

hawaiianer, 2015-03-17:
Is it possible to reverse the yaw servo direction?  yaw_control_direction and yaw_direction do not have any effect. It is possible somehow or do I have to go back to multiwii for a tricopter?

Thanks!

Luz

hawaiianer, 2015-03-17:
Found it - Sorry for the trouble! (Servo Tab - Reverse).

hawaiianer, 2015-03-17:
Still not working (as it does in cleanflight). 

Is there a chance to fly a tri with cleanflight soon?

tricopterY, 2015-03-18:
Setting yaw_direction=-1 behaves the same as MultiWii. Viewing your craft from behind, the rear motor pivoting assembly should be installed slightly tilted to the right eg. 5-10 degrees or say on a face of a clock at 1:00. You should not reverse the servo in cleanflight. May be you should reverse your Tx-Yaw channel. A left stick input should tilt the tail assembly to the right and vice versa, when viewed from behind.

closed 2015-03-21 by hydra