cleanflight/cleanflight #648: Port refactored beeper code from Baseflight

thenickdude, 2015-03-19:
Creating this issue to help track issues that are blocked waiting for beeper code to be ported over from upstream.

Related issues: 

#647 Add beep when changing flight modes
#481 Morse code beeps for failure states
#408 CLI command to play/test warning tones
#317 Beep to sound on arm/disarm

hydra, 2015-03-19:
Thanks for opening this nick.  Much appreciated.

ethomas997, 2015-03-22:
Is anyone currently working on this task?  If not then I'll work on it and post a PR.

--ET

thenickdude, 2015-03-22:
@ethomas997 haven't heard anyone mention it on the IRC channel (I'm not working on it, at least).

hydra, 2015-03-22:
@ethomas997 please go ahead and do this! Many thanks.

ethomas997, 2015-03-25:
Alright, PR is posted.  http://github.com/cleanflight/cleanflight/pull/669
I've been flying this version with no issues.

--ET

hydra, 2015-03-25:
Nice!  I'll check this out.

TadasSu, 2015-03-29:
Hi All.
All beeps are very quiet in CleanFlight (in comparison with baseflight). 
It seems that beeps in BF are way longer (maybe double the time).
Low Voltage beeps almost can not be heard (or hear the difference between low and critical alarm) when flying.
Is it possible to make them louder (longer)?

dkisselev, 2015-03-29:
@TadasSu Have you tried the new beeper code in #669? It'll be in CF 1.9.x and some of the sounds have been changed to be louder/closer to baseflight sounds.

ethomas997, 2015-03-30:
If anyone wants to try out a test hex file I've posted one here:  http://www.etheli.com/CF/refactoredBeeperCode

--ET

TadasSu, 2015-03-30:
Thanks, I will try it :)

ethomas997, 2015-04-20:
Updated PR for this is here:  Port of refactored beeper code 181_1 #730
As far as I know it's ready for merging.

--ET

hydra, 2015-04-28:
done

closed 2015-04-28 by hydra