cleanflight/cleanflight #653: Change buzzer pin to RC_Output_6 on CC3D

digitalentity, 2015-03-21:
According to docs on CC3D pin 6 of RC_Output connector is always unused. Also, buzzer uses simple GPIO bit banging. Why don't use pin 6 of RC_Output to drive an external buzzer?

digitalentity, 2015-03-21:
This will allow us to attach a beeper to CC3D without precise soldering.

Pierre-A, 2015-05-02:
refered in issue ≠832

hydra, 2015-05-15:
See #357 and #884.

If the pin is indeed always unused then it would certainly be a good candidate, yes.  Someone verify this and submit a PR please.

Pierre-A, 2015-05-15:
PR for buzzer on CC3D output 6 is on the way

digitalentity, 2015-05-15:
Oops, sorry @Pierre-A, didn't see you comment.

digitalentity, 2015-05-15:
@hydra, I've been flying with changes from b9f281fdc5bab98e46cddad703a9f517497721cf for a while - no problem so far. The next one 53ce4b224d130a437fc5f5be9ffcb3882a52903b is untested.

Pierre-A, 2015-05-15:
no problem **digitalentity**, so it's done !

closed 2017-02-25 by hydra