cleanflight/cleanflight #654: Yaw action on Tri-copter at Harakiri is crazy

j0uni, 2015-03-21:
I have had problem with harakiri PID on my Tricopter, the Yaw RC-command is so powerful that it's difficult to fly. My yaw-rate is 0. I've tried expo on radios but it just becomes binary-type control. Dual rates don't work either because then you cannot arm the copter - and loose the resolution.

I've already patched my own fork of the Harakiri PID by just dividing the YAW RC command half. It is then really close the other PID yaw action. 

I can make pull request if this feature is wanted. If it reduces the yaw too much, there's always possibility to increment the yaw rate and get the gain back.

thenickdude, 2015-03-21:
I can't think of a good reason why the yaw rate would have a different effect on a tricopter, as it is supposed to control the rotation rate as measured by the gyroscopes (so if a tricopter had a "stronger yaw system", it should be detected as yaw rate overshoot and the control signal should weaken). I think discovering the reason for the difference would be useful.

j0uni, 2015-03-21:
On other PIDs the yaw is working "normally". My friend said that the yaw is also very powerful on his quad (compared to other PIDs) but it's flyable. So clearly there's something different in calculations.

ledvinap, 2015-03-21:
@thenickdude : Maybe the pid is saturated for quad (or yaw jump limit is causing saturation?)

hydra, 2015-03-25:
I found my CJMCU micro quad unflyable with Harakiri.  I need to give this a test.

@j0uni Can you give us the spec of your quad and a config dump?

j0uni, 2015-03-26:
Sure @hydra. My tricopter is 210mm M2M and I'm running 5" props (3blade) on 3S pack at 1900kv. The servo is directly connected to yaw axis (no linkage). Pic here: http://tulos.tuu.fi/wp-content/uploads/2015/02/IMG_4213.jpg

Settings here: http://tinyurl.com/pmjbwub

Try it, it's really "natural" with my fix.

hydra, 2015-05-15:
#657 was merged.  Please re-test.

j0uni, 2015-05-15:
@hydra merged? i cant see it merged. I currently havent got tri to test atm.

closed 2017-02-25 by hydra