cleanflight/cleanflight #660: Can we consider using a RTOS for cleanflight?

digitalentity, 2015-03-22:
I was thinking, CF is running a single time-critical task - gyro & acc stabilization. All other stuff (rc input, logging, GPS, althold, poshold, beeping, ledstrip, telemetry, msp etc.) is in fact asynchronous to main stabilization loop. Why not use some RTOS (FreeRTOS?) to do the scheduling?
This will eliminate issues like buzzer blocking the main loop, allow us to run heavy and time-consuming navigation tasks without struggling to keep looptime within reasonable limits.
Also, moving to RTOS will greatly simplify a watchdog (#617) implementation as only the time-critical main loop should be monitored to keep drone in the air.

thenickdude, 2015-03-22:
The buzzer doesn't block the main loop, the only buzzer related routine that does block is the one that flashes the lights at the same time, which is easily replaced with more of the non-blocking stuff that beeper.c does already.

Right now there's nothing so slow that is preventing us from running nice looptimes. The main advantage would be for new features that are added that are so slow that they take multiple looptimes to run a task, and we don't want to have to write our own code to break the task up into chunks. I don't think navigation is one of those, TBH.

digitalentity, 2015-03-22:
I think althold and poshold should be recalculated on any new data, that would be baro or sonar for althold, and gps update for poshold. Both happen asynchronous from main loop. RTOS would allow us to simply run a task without thinking about how often we should call some function from the main loop. That would greatly improve code readability.

thenickdude, 2015-03-22:
True!

enly1, 2015-03-22:
If this goes beyond a discussion, for STM32 processors, Chibios support is
excellent. www.chibios.org

hydra, 2015-03-22:
I was having discussions regarding this exact topic in the pub in mayrhofen on friday!  At some point yes it would be nice to look into this.

ledvinap, 2015-03-22:
I am not that sure that switching to RTOS will be worth it. 
- There are other firmwares that use RTOS, it may be easier to port CF features to them.
- Using RTOS will create new problems - resource locking, priority handling, race conditions. Some bugs may be considerably harder to find in comparison with current model.

digitalentity, 2015-03-22:
I agree with @hydra, switching to RTOS now wouldn't be

tricopterY, 2015-03-23:
:+1: @ledvinap

ledvinap, 2015-03-23:
BTW: Maybe it would be interesting to connect CF and Intel Edison. It may provide best of both worlds..

tricopterY, 2015-03-23:
@ledvinap - it does not add up. What do you mean?

ledvinap, 2015-03-23:
Keep CF, but allow sending all sensor, RC, GPS data over serial and receive motor commands back.
It should be quite easy to move current loop internals to linux thread.
Also CF will work as failsafe, just implement this as 'REMOTE' mode ...

hydra, 2015-03-28:
Better task scheduling instead of task jitter the main issue that needs to be solved, currently there are some issues that would benefit from this.

1. GPS SV Frame processing - it's intermittent when unarmed with OLED display, every so often the timers for the OLED and GPS get out of phase and sometimes GPS frames are not received correctly and then are eventually received properly when the phase changes.
2. LED and LED strip flashing.  Sometimes the flashes are too short, sometimes too long.
3. Beeper, same issue as LED flashing.
4. Calibration of sensors is affected by loop time/jitter.

kylemanna, 2015-03-28:
It's worth mentioning that [Pixhawk runs NuttX](https://pixhawk.org/dev/nuttx/start) RTOS.

closed 2017-02-25 by hydra