cleanflight/cleanflight #662: Enhancement : autotune documentation

rocky77, 2015-03-23:
I think that the sentence "Autotune does not support pid_controller 2 (pid_controller 0 is the Cleanflight default)."  could be change by "Autotune only support pid_controller 0 (pid_controller 0 is the Cleanflight default)."

thenickdude, 2015-03-23:
That's not true though, it supports all the PID controllers that use integral PID settings, which is every controller except number 2.

rocky77, 2015-03-23:
OK,  I didn't know that. So we can use Autotune with all  pid_controller except number 2, is it true?

So we can change for better understanding,  by "Autotune supports all the Pid controllers that use integral Pid settings except pid_controller 2 (pid_controller 0 is the Cleanflight default)."?

thenickdude, 2015-03-23:
I don't think that would be helpful because there's no reason to expect people to know which PID controllers use integral PIDs (it's an implementation detail that pretty much has no visible impact).

hydra, 2015-03-25:
Closing.  Nothing to do IMHO.

closed 2015-03-25 by hydra