cleanflight/cleanflight #664: Serial RX and GPS apparently invalid

hodaddy4, 2015-03-24:
I am running a cc3d flashed to clean flight and spectrum satellite on UART1 as RX_SERIAL and UART3 sending CLI, MSP, GPS Pass thru to Bluetooth and odd. I enable softserial and everything is fine when I assign GPS to SOFTSERIAL1. When I enable GPS the board resets and all my port settings are back to default. Is this a bug or am I doing something invalid?

thenickdude, 2015-03-24:
Which firmware and configurator version? (both bits of info are printed in the configurator log at the top of the window)

hodaddy4, 2015-03-24:
Clean flight 1.7.2 and configurator 0.62.1

hodaddy4, 2015-03-24:
If it makes a difference, I did the full flash (not the dual boot loader)

hodaddy4, 2015-03-24:
As an update to this, I flashed Clean Flight 1.8 and used the 0.63 Configurator. It doesnt crash when assigning the com ports as previously stated, but no GPS was found. I reconfigured the ports to try and use MSP on SOFTSERIAL. It 'kind of' connected  to configurator but was very slow and intermittent. sending commands on CLI echoed missing letters of each command. Started adjusting the settings of GPS to slow it down. I set the GPS according to the GPS file in the docs section. I ended up changing the NEO-6M from 10Hz to 5Hz and started at 9600 baud. Worked... 38400, worked... 57600, worked... 115200, didn't even show GPS connected. Rolled the GPS back to 57,600 at 5hz and all seems well. I know 1.8 isn't labeled as stable release yet so I'm hoping I won't have any actual flight issues. Maybe all this info helps, maybe not. Either way, It's what I did get GPS on SOFTSERIAL and RX_SERIAL to work on UART1, leaving UART3 available for bluetooth/data modem to ground station.

digitalentity, 2015-03-24:
I can confirm that MSP on softserial is slow, cli sometimes shows corrupted
and missing letters. Don't have my logic probe at the moment, so I can't
chuck what's actually happening on the wire.

hydra, 2015-03-25:
There is work in progress code changes to improve softserial.  See #113.

With regard to 1.7.2 that is expected behaviour if your configuration is invalid.  You probably forgot to configure something.

closed 2015-03-25 by hydra